dsmrreader/dsmr-reader #1393: MQTT: Fix support for QoS 0

dennissiemensma, 2021-05-17:
- #1391
- #1387

dennissiemensma, 2021-05-17:
- Onderzoeken of de eerste `client.loop()` zonder return-check gedaan kan worden. Voor QoS 1/2 is de `is_published()` check er namelijk alsnog daaronder.
- MQTT client timeout flink verlagen.

bockelaar, 2021-05-25:
ik heb ook last van deze bug. na verloop van tijd (paar uur) stopt de verwerking. een sudo supervisorctl restart all maakt dat alle achterstallige telegrams etc weer verwerkt worden.
Als je info/logs e.d. nodig hebt laat het me maar weten

dennissiemensma, 2021-05-25:
@bockelaar bedankt voor je melding. Voor nu kun je het beste QoS 2 gebruiken. Dat zou het ook moeten oplossen.

bockelaar, 2021-05-25:
ik las het, het het een uurtje geleden ingesteld :)

dennissiemensma, 2021-05-30:
Ik zal ervoor zorgen dat DSMR-reader voor nu alleen QoS 2 gebruikt. Ook automatisch voor iedereen die nog 0 of 1 gebruikt.

Dan is het ook geen variabele instelling meer en kan er minder stuk gaan. Ik heb ook tot nu toe nog geen meldingen gehad dat QoS 2 niet goed werkt.

Ik zal het eerst hardcoden en de instelling verbergen in een patch release. Dan gooi ik het er later wel helemaal uit.

closed 2021-05-30 by dennissiemensma