dsmrreader/dsmr-reader #1395: CSV export

jjvereijken, 2021-05-25:
Als ik via POSTs naar api/v2/statistics/day historische data importeer, lijkt alles perfect te werken, maar kan ik daarna geen CSV exports meer maken. In het "Export data" scherm zijn er helemaal geen data meer te selecteren, en als je toch op "Download export" duwt, dan krijg je een lege CSV file.

Ik had eerst twee jaar aan oude data geïmporteerd, zag er allemaal super uit, maar na een tijdje zag ik dit probleem. Na terugzetten van de database en opnieuw proberen zag ik dat het al misgaat als je één dag aan oude data inschiet. Ik heb gezorgd dat de historische data exact aansluiten (geen "gaten" in de datumreeks).

Reproduceren:

- Ga naar "Export data", selecteer data range, doe "Download export" -> alles werkt
- Importeer één dag aan historisch data, direct voor de huidige meetreeks, met een POST naar api/v2/statistics/day
- Ga naar "Export data" -> niet mogelijk om een data range te selecteren

Ik gebruik DSMR-reader v4.16.2 in de standard Docker set-up op een RaspBerry Pi 3B onder Ubuntu 20.04, en alles lijkt verder perfect te werken.

Vergeet ik hier iets simpels, of is dit een bug?

dennissiemensma, 2021-05-25:
Bedankt voor je melding! Ik denk dat ik het probleem al zie in de Export-pagina. 

Voor de start- en einddatum pakt die de eerste en laatste dag, maar op basis van ID ipv datum. Dat gaat goed, totdat iemand een import doet waardoor de ID's afwijken van de datums. Dus een lelijke bug en ik zal het meenemen bij de volgende release.

Het lijkt er overigens al 5 jaar in te zitten, maar je moet er net tegenaan lopen denk ik. :smile: 

[image]

jjvereijken, 2021-05-26:
Super, dankjewel!

Ik heb om een andere oorzaak uit te sluiten al mijn dagtotalen uit het api/v2/statistics/day eindpunt getrokken, de ID's verwijderd, een schone lege install gemaakt, en alles weer op de correcte datumvolgorde ingeschoten. Daarna kon ik idd weer keurig CSV exports maken (en alle andere dingen werkten ook prima)!

dennissiemensma, 2021-05-26:
Bedankt voor je update! Dat bevestigt mijn vermoeden. Het is uiteindelijk een kleine fix, maar fijn dat je voor jezelf al een snellere workaround hebt gevonden!

dennissiemensma, 2021-05-30:
Gaat mee in v4.17

closed 2021-05-30 by dennissiemensma

dennissiemensma, 2021-09-19:
Gefixt in [v4.17](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.17.0)