dsmrreader/dsmr-reader #1396: Laagtarief en hoogtarief zijn omgedraaid

jimmyd-be, 2021-05-27:
Ik gebruik al enkele weken de prachtige applicatie. Het enige wat opviel is dat het laagtarief overdag gezien wordt en hoogtarief in de nacht. Volgens mij moet dit omgekeerd zijn. Is dit ergens in te stellen of is dit een fout?

Ik gebruik de remote datalogger om data naar mijn server te sturen. Ik zit in België op de meter van Fluvius.

```

DSMR-READER
    App / Python / Database                                                     v4.16.2 / v3.9.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                             "None" / "101"

DATA
    Telegrams total (est.)                                                                            1157386
    Consumption records electricity / gas (est.)                                                 40318 / 8563

```

dennissiemensma, 2021-05-27:
Bedankt voor je melding. Het DSMR-protocol kent geen hoogtarief of laagtarief, alleen tarief `1` of `2`. De betekenis van elk tarief kan verschillen. In Nederland is het laag- en hoogtarief, in België dag- en nachttarief.

Daarom gebruikt DSMR-reader standaard de Nederlandse namen, maar je kunt deze zelf anders noemen en dus ook omdraaien. Zie #750 voor meer informatie.

dennissiemensma, 2021-05-27:
Zie ook #858 en je kunt het instellen binnen DSMR-reader op de URL: `/admin/dsmr_frontend/frontendsettings/1/change/`

closed 2021-05-27 by dennissiemensma

jimmyd-be, 2021-05-27:
Bedankt deze setting had ik nog niet gevonden. Bedankt voor het snelle antwoord.