dsmrreader/dsmr-reader #1397: Day statistics are lagging behind

digiater, 2021-05-28:
DSMR-reader v4.16.2

De volgende melding staat op de 'about' pagina:

      Day statistics are lagging behind   (1 day, 9 hours ago)

Ik moet vermelden dat gisteren de gasmeter is verwijderd omdat we van het gas af zijn. En dat de update van die dag dus ergens fout loopt, waarschijnlijk omdat er geen gas gegevens meer worden doorgegeven.

In de logs directory van dsmr-reader staat alleen oude data van vorig jaar.

Wat kan er aande hand zijn?

Jur.

dennissiemensma, 2021-05-28:
Bedankt voor je melding. De kans is groot dat het inderdaad ligt aan dat de gasgegevens "opeens" ontbreken voor DSMR-reader. 

Deze detecteert helaas niet automatisch dat de meter uberhaupt weg is, maar ik denk dat je het wel makkelijk kan fixen via een vinkje in de admin. 
Bij de backend-instellingen op de pagina `/admin/dsmr_backend/backendsettings/` kun je een vinkje aanzetten waarmee gas-verbruik genegeerd wordt. Ik hoor graag of dat het voor je oplost.

digiater, 2021-05-29:
Dat lost het inderdaad op. Bedankt!

Jur.

closed 2021-05-29 by digiater