dsmrreader/dsmr-reader #1398: met foute mqtt dsmrreader stukgemaakt?

migimli, 2021-05-29:
hi,

ik had al een tijd geprobeerd om mqtt naar homeassistant te doen, lijkt dat ik een fout heb gemaakt ergens.
mis nu sinds 21 mei mijn gasmetingen in het systeem. stroom werkt nog wel

hoe kan ik dit slim weer terugzetten? een backup maken in nieuw systeem lijke niet slim, hou ik zelfde data?

DSMR-READER
    App / Python / Database                                                     v4.16.2 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             303255
    Consumption records electricity / gas (est.)                                                 45363 / 2939

UNRESOLVED ISSUES
    Proces loopt achter op schema: Weather update                                          19 minuten geleden
    Proces loopt achter op schema: Generate day and hour statistics                        53 minuten geleden
    Proces loopt achter op schema: Generate consumption data                               58 minuten geleden
    Proces loopt achter op schema: Retention data rotation                                     1 week geleden
    Te veel onverwerkte telegrammen: 66767                                                     1 week geleden
    Dagstatistieken lopen achter                                                        1 week, 1 dag geleden

```

heb alvast mijn pi en de app helemaal updated.

dennissiemensma, 2021-05-29:
Bedankt voor je vraag. Ik denk dat je het beste debuglogging aan aanzetten om te zien waar het op wacht: https://github.com/dsmrreader/dsmr-reader/issues/1387#issuecomment-839932356

Je zou dan in de log van `dsmr_backend` wel een patroon moeten zien van een fout of iets wat lang lijkt te duren.

dennissiemensma, 2021-05-29:
Het kan trouwens zijn dat je MQTT met QoS 2 in DSMR-reader moet draaien. Daar gaat toevallig dat issue #1387 (maar ook #1391) over, al hoeft het niet dezelfde oorzaak te zijn.

migimli, 2021-05-29:
rare is dat ik nu mijn gasmetingen volledig kwijt ben, maar stroom nog wel zie? kan dat een verband hebben?

migimli, 2021-05-29:
geloof dat ik begin in te lopen nu ik de boel stilzette en QoS2 heb ingesteld:)

dennissiemensma, 2021-05-29:
Het kan even duren voordat alles is ingehaald:

```
Te veel onverwerkte telegrammen: 66767 
```

migimli, 2021-05-30:
dank! werkt helemaal nu, ipv mqtt naar homeassistant maar influx koppeling proberen te maken, :)
daar nieuwe leermomenten in aantocht, 

2021-05-30 12:40:09,914 | ERROR    services.run:75 | INFLUXDB: Writing measureme                                                                 nt(s) failed: 400: {"error":"unable to parse 'electricity_voltage 16223712050000                                                                 00000': invalid field format"}
maar ga ik eerst zelf puzzelen. deze dicht denk ik!

closed 2021-05-30 by migimli