dsmrreader/dsmr-reader #1399: [FR] Opbrengst in euro's voor teruglevering laag/hoog en totaal ook zichtbaar maken.

E020, 2021-05-30:
Zou fijn zijn om de de kosten/opbrengst in Euro's te splitsen naar afgenomen/teruggeleverd. Nu zie ik alleen eind kosten en niet de splitsing. Of kijk ik niet goed? ;-)

dennissiemensma, 2021-05-30:
Bedankt voor je vraag. De reden dat teruglevering niet expliciet vermeld wordt, is omdat het vrij complex is qua saldering. 

Daarom wordt het niet getoond om te voorkomen dat er verkeerde conclusies getrokken worden door gebruikers. Zelfs de huidige implementatie is geen garantie dat het spiegelt met de daadwerkelijke afrekening. DSMR-reader kan alleen wat zeggen over het verbruik en een schatting wat "een dag kost".

E020, 2021-05-30:
ok helder. Maar de kosten zijn met zonnepanelen al (vaak) negatief bij mij dus uiteindelijk vind de berekening al plaats. Dus was benieuwd. Maar ga kijken of ik het via home assistant zelf kan berekenen. Thx.

dennissiemensma, 2021-05-30:
Prima! Daar doel ik overigens op. Want dat DSMR-reader het op die manier berekent, betekent niet dat dat goed spiegelt met de afrekening van je leverancier. Dus wees bewust van eventuele afwijkingen.

Want in de meeste gevallen is er voor een contractperiode (of kalenderjaar?) een max kWh wat gesaldeerd kan worden. Mocht je daar bijvoorbeeld over heen komen dan klopt DSMR-reader al helemaal niet, omdat je dan daarna een lagere prijs krijgt.

closed 2021-05-30 by dennissiemensma