dsmrreader/dsmr-reader #1400: Hi,

hemant5400z, 2021-06-01:
Weet iemand wat de beste manier is om de alle data te exporteren naar influx. Ik heb het nu een aantal keer gedaan:

Influx db aangemaakt
Export gestart via cmnd van dsmr met batches.

Als ik klaar is gooi ik de logger weer om naar de zo net gemaakte influx db. Echter nu zie ik telkens record already exist of type floting point en dropped en dus krijg ik geen nieuwe records binnnen.

Doe ik export laatste 50 dan is ie weer up to date maar nieuwe krijg ik helaas niet binnen automatisch.

> 🇳🇱 Lees eerst [deze info](https://github.com/dsmrreader/dsmr-reader/blob/v4/CONTRIBUTING.md) om jezelf te helpen.
> 
> 🇬🇧 First read [this info](https://github.com/dsmrreader/dsmr-reader/blob/v4/CONTRIBUTING.md) for self-service.

dennissiemensma, 2021-06-01:
Bedankt voor je melding. Ik denk dat je het beste kunt aangeven wat je exact uitvoert en welke foutmeldingen je precies krijgt.

hemant5400z, 2021-06-02:
Hoi,

ik heb het op 2 manier geprobeert:

Lege database.

Manier 1:
./manage.py dsmr_influxdb_export_all_readings --to-influx-database dsmrreader-export --max-batches 11000

Bovenstaande opdracht is in 2 uur klaar en de database heeft dan alle data, echter zodra ik dan DSMR influx config omzet naar deze database krijg ik de melding dat het record als bestaat en dus krijg ik de nieuwe data niet in de database.

Manier 2:

Nieuwe database en logger wijst al naar deze data zodat ik nieuwe gegevens binnen krijg.
Op de achtegrond voer ik weer uit:

./manage.py dsmr_influxdb_export_all_readings --to-influx-database dsmrreader-export --max-batches 10999 (1 minder dan laatste batch.

En ik zie dat de database wordt gevuld en begint bij de oudste gevens echter zodra hij aan het eind komt krijg ik de volgende melding:
2021-06-02 09:31:42,558] INFO     INFLUXDB EXPORT: Last PK 965300 @ 2021-06-01T23:48:01.923597Z (exp
ort_progress_meta in InfluxDB)                                                                       
[2021-06-02 09:31:42,559] INFO     INFLUXDB EXPORT: Limiting to 10 batches of max 100 readings each  
[2021-06-02 09:31:42,560] INFO     INFLUXDB EXPORT: Fetching readings with PK > 965300 (max number of
 readings 1000)                                                                                      
[2021-06-02 09:31:56,754] INFO     INFLUXDB EXPORT: Processing batch (limit: 10): 1 of 10            
[2021-06-02 09:31:59,742] ERROR    INFLUXDB EXPORT: Writing measurement(s) failed: 400: {"error":"par
tial write: field type conflict: input field \"returned_2\" on measurement \"electricity_positions\" 
is type float, already exists as type integer dropped=108"}                                          
Traceback (most recent call last):                                                                   
  File "/dsmr/./manage.py", line 10, in <module>                                                     
    execute_from_command_line(sys.argv)                                                              
  File "/usr/local/lib/python3.9/site-packages/django/core/management/__init__.py", line 401, in exec
ute_from_command_line

En dan mis ik dus de laatse 2 dagen als ik de influx db de laatse 2 dagen laat droppen dan ik het zelfde probleem.

Hemant

hemant5400z, 2021-06-02:
Als ik de influx config tijdelijk omzet naar een andere database dmsr_temp en vervolgens alles importeer naar dmsr, dan kan ik alle data zien als ie klaar is na controle zet ik de database weer om naar de oorspronklijke.

Echter zie ik dan in de dmsr console constant in mijn log het volgende:

2021-06-02 16:16:36,798 DEBG 'dsmr_backend' stdout output:                                           
[2021-06-02 16:16:36,798] ERROR    INFLUXDB: Writing measurement(s) failed: 400: {"error":"partial wr
ite: field type conflict: input field \"current_l1\" on measurement \"electricity_power\" is type flo
at, already exists as type integer dropped=1"}                                                       
                                                                                                     
2021-06-02 16:16:36,813 DEBG 'dsmr_backend' stdout output:                                           
[2021-06-02 16:16:36,813] ERROR    INFLUXDB: Writing measurement(s) failed: 400: {"error":"partial wr
ite: field type conflict: input field \"returned_2\" on measurement \"electricity_positions\" is type
 integer, already exists as type float dropped=1"}                                                   
                                                                                                     
2021-06-02 16:16:47,921 DEBG 'dsmr_backend' stdout output:                                           
[2021-06-02 16:16:47,921] ERROR    INFLUXDB: Writing measurement(s) failed: 400: {"error":"partial wr
ite: field type conflict: input field \"current_l1\" on measurement \"electricity_power\" is type flo
at, already exists as type integer dropped=1"}                                                       
                                                                                                     
2021-06-02 16:16:47,938 DEBG 'dsmr_backend' stdout output:                                           
[2021-06-02 16:16:47,938] ERROR    INFLUXDB: Writing measurement(s) failed: 400: {"error":"partial wr
ite: field type conflict: input field \"returned_2\" on measurement \"electricity_positions\" is type
 integer, already exists as type float dropped=1"}                                                   
                                                                        
Hemant

dennissiemensma, 2021-06-02:
Bedankt voor je aanvulling. Ik zal er een keer wat dieper naar kijken of ik dezelfde fouten krijg.

Welke versie van InfluxDB gebruik je?

hemant5400z, 2021-06-02:
Ik gebruik op dit moment    1.8 docker versie

hemant5400z, 2021-06-03:
Hoi,

Ik merk nu dat de live data na de foutmelding niet meer wordt geladen.

Echter als ik:
./manage.py dsmr_influxdb_export_all_readings --to-influx-database dsmrreader-export --max-batches

Dan is is weer up-to date maar er komt niks nieuws bij.

dennissiemensma, 2021-06-03:
Je kunt sowieso het beste dan in de backend-logs kijken voor fouten. En anders debug-logging tijdelijk aanzetten om te kijken waarom de backend stopt met verwerken.

Zie: https://github.com/dsmrreader/dsmr-reader/issues/1387#issuecomment-839932356

hemant5400z, 2021-06-04:
Hoi Dennis,

De Fouten zijn duidelijk hij verwacht voor een aantal velden ineens een anders formaat. Dat heb ik nu opgelost door in DSMR de Influx velden die problemen gaven te hernoemen. En dat loopt de Live data gewoon door.

Waarom hij ineens een ander formaat (zie foutmelding) verwacht is me niet uidelijk.

Uiteindelijk ziet het er zo uit in Influx.

[image]

dennissiemensma, 2021-06-04:
Bedankt voor je aanvulling. Goed om te weten dan. Wellicht is de veld-mapping tussen de eenmalige import en vervolgens de stroom aan nieuwe metingen toch niet helemaal compatible met elkaar.

dennissiemensma, 2021-06-04:
> Ik gebruik op dit moment 1.8 docker versie.
> 
> Versie 2 wordt niet gesupport althans kon de settings niet kwijt.

Nog in aanvulling hierop: Klopt. Dit wordt pas mogelijk in DSMR-reader v5.x. Zie #1380.

dennissiemensma, 2021-09-22:
Dit wordt veroorzaakt doordat de eenmalige export de velden als interne decimals, dus kommagetallen, exporteert.

Bij de andere ingebouwde export, direct na elke meting, wordt het eerst tijdelijk omgezet naar een JSON-structuur, waar kommagetallen zonder decimalen (zoals `0,0`) als bijeffect omgezet worden naar `0`. Dit is voor InfluxDB een ander type en daar komt die foutmelding uit.

Oplossing wordt om te zorgen dat DSMR-reader het op een andere manier exporteert (zonder JSON-conversie), waardoor het gewoon kommagetallen blijven en er later geen conflicten ontstaan.

dennissiemensma, 2021-09-22:
Ik heb er een fix voor in release 1.8. Die is echter niet compatible met eventuele data die gequeued is om naar influx db te gaan. Dat lijkt verder geen impact te hebben, omdat een of meerdere metingen missen doorgaans geen invloed heeft op rapportages.

Desondanks zal ik een waarschuwing in de changelog zetten.

closed 2021-09-22 by dennissiemensma

dennissiemensma, 2021-09-29:
Uitgebracht in [v4.18](https://github.com/dsmrreader/dsmr-reader/discussions/1456)