dsmrreader/dsmr-reader #1401: DJANGO_FORCE_SCRIPT_NAME env value wordt niet gebruikt in API docs

swerveshot, 2021-06-01:
### Wat gebruik je?
* DSMR-reader version: `v4.11`
* Hardware: `Odroid U3; ARMv7`
* DSMR protocol: `v5`
* OS: `Ubuntu 16.04 LTS`

### Omschrijving
De DJANGO_FORCE_SCRIPT_NAME omgevingsvariabele lijkt niet gebruikt te worden voor de API docs.

### Reproduceren issue
Doe een install van v4 en gebruik de in #1140 geïntroduceerde variablen. Bijvoorbeeld:
```
DJANGO_STATIC_URL="/dsmrreader/static/"
DJANGO_FORCE_SCRIPT_NAME="/dsmrreader"
DJANGO_USE_X_FORWARDED_HOST=true
```
#### Voorbeelden
* De link naar de Postman collection is nu `http://myhost/static/dsmr_frontend/postman/collection.json` terwijl deze in werkelijkheid hier staat `http://myhost/dsmrreader/static/dsmr_frontend/postman/collection.json`.
* De gegenereerde URL voor live electricity consumption is `http://myhost/api/v2/consumption/electricity-live` in plaats van `http://myhost/dsmrreader/api/v2/consumption/electricity-live`.

### Wat werkt wel
* ReDoc pagina's; Changelog, Postman collection, api.
* Link naar OpenAPI specification

### Wat werkt niet
* Links naar de Postman collection and variabelen
* Alle gegenereerde copy-pastebare URLs in de voorbeelden

### Mogelijke oplossingen
* Gevalletje PEBKAC; configuratiefout. Ik heb al de nodige testen gedaan, maar zonder resultaat. Upgraden naar hoger dan v4.11 heb ik geprobeerd, maar dat krijg ik vanwege m'n antieke distro helaas niet voor elkaar.
* Als er geen mogelijkheden zijn om ReDoc deze variable te laten gebruiken houdt het op natuurlijk.

dennissiemensma, 2021-06-01:
Bedankt voor je melding. Het is mij niet helemaal helder waar dat aan ligt. Sowieso heb ik zelf geen ervaring met die specifieke instellingen om het in Django werkend te krijgen.

- De URL's in de OpenAPI specs komen uit Django REST Framework. Wellicht houdt die geen rekening met niet-standaard instellingen, want ik zie de "normale" URL's in de JSON staan, zonder de prefix. Ik weet niet of het uberhaupt kan fixen.
- De URL's [naar Postman zijn hardcoded](https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmrreader/urls.py#L13-L14), dus dat verklaart waarom de prefix daar mist. Ik zou daar nog wel een lazy-lookup kunnen doen, zodat die de URL genereert (hopelijk inclusief prefix) ipv hardcoded.

Wellicht ben je sowieso het meeste geholpen met gewoon [een directe download vanuit Github](https://github.com/dsmrreader/dsmr-reader/tree/v4/dsmr_frontend/static/dsmr_frontend/postman) en dan kun je de docs er naast leggen ter referentie. 
In Postman kun je dan in de `{{DSMR-reader-API-host}}` env variabele je prefix uit `DJANGO_FORCE_SCRIPT_NAME` achter de hostname instellen en dan hoef je niet alle URL's handmatig aan te passen in Postman (of ze te copy pasten en dan aan te passen).

Wellicht niet helemaal perfect, maar dan kun je iig wat makkelijker ontwikkelen tegen de API. Het eerste puntje kan ik denk ik niet fixen, dan zou ik in dat API Framework moeten duiken wat die onderwater doet en of er een workaround is.

swerveshot, 2021-06-02:
Het is ook absoluut een minor ding hoor. Maar ik dacht; als het een easy fix is, is het mooi meegenomen. Maar als er serieus werk in gaat zitten never mind dan. Het is niet alsof ik vergeet dat er een subpath tussen het request URL moet komen als ik dat zelf geconfigureerd heb. 😄 

Goed werk met die env variabelen trouwens. Vanwege een upgrade die de soep in was gedraaid heb ik alles opnieuw geïnstalleerd en alle 'tweaks' die ik in het verleden moest doen zitten nu in variablen. Ideaal!

dennissiemensma, 2021-06-02:
Bedankt voor je aanvulling. In dat geval zal ik later een keertje kijken of ik iig die Postman-files goed krijg qua URL.

dennissiemensma, 2021-09-20:
In v4.18 zit straks een mogelijke fix, die simpelweg `DJANGO_STATIC_URL` (= `STATIC_URL` intern) als basispad gebruikt. 

Gezien die i.c.m. `DJANGO_FORCE_SCRIPT_NAME` ook gezet hoort te worden, zou die overeen moeten komen waardoor het basis pad weer "klopt".

closed 2021-09-20 by dennissiemensma

dennissiemensma, 2021-09-29:
Uitgebracht in [v4.18](https://github.com/dsmrreader/dsmr-reader/discussions/1456)