dsmrreader/dsmr-reader #1402: Onverwerkte telegrammen na upgrade

reboot1983, 2021-06-01:
Sinds de upgrade van een uur geleden van 4.16.2 naar de laatste versie (4.16.3) krijg ik een dsmr reader melding dat hij onverwerkte telegrammen heeft. En dat getal loopt op, de grafieken staan stil, maar het actuele verbruik lijkt wel gewoon te kloppen.

hoe krijg ik dsmr reader weer aan de praat dat ie de onverwerkte telegrammen weer gaat verwerken? Het betreft een dedicated install op Debian conform de v4 handleiding, gekoppeld aan de p1reader (ethernet versie) van Zuidwijk.com. 

“supervisorctl restart all” helpt niet. Postgresql opnieuw starten helpt niet. System reboot helpt niet. Wat kan ik nog meer proberen?

Ik kan in de changelog of handleidingen hier ook geen stappenplan voor vinden lijkt het?

Alvast bedankt voor de tips/tricks.

[image]

dennissiemensma, 2021-06-01:
Je kunt het beste eerst zoeken in de logfile naar de oorzaak: https://github.com/dsmrreader/dsmr-reader/issues/1387#issuecomment-839932356

reboot1983, 2021-06-02:
Ik zie in de dsmr_backend log dat er issues waren om te verbinden met postgresql.
Maar die issues bleven ook nadat postgresql herstart was en weer gewoon up en running en beschikbaar was.

Ik snap alleen niet waarom dit niet automatisch is opgelost met een supervisor restart all. Ik heb nu specifiek en alleen de datalogger herstart en daarmee heeft ie alles in een paar tellen verwerkt en nu draait ie weer goed.

the1rig, 2021-06-02:
Ik merk hetzelfde, zelfde foutmelding ook. Alleen lost het opnieuw opstarten van het dsmr_datalogger niet op.

dennissiemensma, 2021-06-02:
@reboot1983 bedankt voor de aanvulling. Werkt het inmiddels nu wel goed?

Het lijkt erop dat deze fouten specifiek komen door het herstarten van de database:
```
terminating connection due to administrator command
```

- Waren deze fouten er eerder ook al, voordat je de database herstart hebt?
- En gebruik je toevallig MQTT in DSMR-reader? Want de patch update die je gedaan hebt had alleen invloed op MQTT dacht ik.

dennissiemensma, 2021-06-02:
@the1rig welke foutmelding krijg je? En wat zeggen de logs? Wellicht is het sowieso handig om even een los issue aan te maken hier. Je kunt de huidige uiteraard watchen.

the1rig, 2021-06-03:
@dennissiemensma het probleem lijkt vanzelf verholpen te zijn, ik had overigens dezelfde foutmelding als @reboot1983, maar na een nacht is hij weer helemaal bij en blijft hij bij :)

dennissiemensma, 2021-06-03:
Oke dank voor de update. Ik zou zeggen houd het voor nu even in de gaten of het vaker voor komt. 
De laatste patch-update betreft namelijk vooral MQTT en *zou* geen andere zaken moeten raken.

closed 2021-06-03 by dennissiemensma

reboot1983, 2021-06-04:
> @reboot1983 bedankt voor de aanvulling. Werkt het inmiddels nu wel goed?
>
> Het lijkt erop dat deze fouten specifiek komen door het herstarten van de database:
>
> ```
> terminating connection due to [...]

Sorry voor de late reactie.

Nee, ik gebruik geen MQTT. Ik moest eerlijk gezegd ook even opzoeken wat het was. 😉 
Na het herstarten van de datalogger werkte het bij mij weer inderdaad. Ik heb voor de herstart van de database (achteraf betrof het een apt upgrade) geen fouten gezien.

Ik zie wel dat de logs vol staan met dezelfde meldingen: 
```Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
```

Dit komt in zowel de backend (paar entries: 99x) als de datalogger (vele entries: 5552x) logfiles veelvuldig voor.
Maar voor de rest is het een oase van rust 😉 

Dankjewel voor de support

dennissiemensma, 2021-06-04:
Bedankt voor de aanvulling. Die `Current logging level set to "ERROR"...` is bedoeld om te laten zien dat de logging werkt, maar ook een hint de huidige instellingen alleen foutmeldingen loggen en geen DEBUG. Als het goed is komt de melding 1x voorbij bij (her)start van een proces. Dit alles om het aantal schrijfacties te beperken.
Als je die 5000x in de datalogger-log ziet, is het wellicht nog de moeite waard om te kijken of dat proces wellicht veel crasht, of dat dat puur een periode was waarin het proces zichzelf steeds herstarte.

Qua logging in de About-pagina, dat is wat lastig omdat strikt gezien DSMR-reader niet perse altijd bij die logs kan. Voor een standaardinstallatie wel, maar het staat iedereen vrij om andere loglocaties te definieren en dat kan buiten DSMR-reader om.