dsmrreader/dsmr-reader #1404: Kan pip versie niet upgraden

Roel0412, 2021-06-03:
Sinds een paar updates van de applicatie kan de PIP niet upgraden naar de nieuwste versie,

WARNING: You are using pip version 21.0.1; however, version 21.1.2 is available.
You should consider upgrading via the '/home/dsmr/.virtualenvs/dsmrreader/bin/python3 -m pip install --upgrade pip' command.

Dit voer ik uit en dan krijg ik dit:

pi@raspberrypi:~ $ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -m pip install --upgrade pip
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (21.0.1)
Collecting pip
  Downloading https://www.piwheels.org/simple/pip/pip-21.1.2-py3-none-any.whl (1.5 MB)
     |████████████████████████████████| 1.5 MB 2.6 MB/s
**ERROR: Will not install to the user site because it will lack sys.path precedence to pip in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages**
WARNING: You are using pip version 21.0.1; however, version 21.1.2 is available.
You should consider upgrading via the '/home/dsmr/.virtualenvs/dsmrreader/bin/python3 -m pip install --upgrade pip' command.

Is hier een oplossing voor?

dennissiemensma, 2021-06-04:
Bedankt voor je melding. DSMR-reader gebruikt actief `pip`, echter `pip` is geen onderdeel van DSMR-reader. Daarom ben je in de meeste gevallen afhankelijk van je OS of Python (hoe het geinstalleerd is). 
In de meeste gevallen wordt er een kopie gemaakt specifiek voor DSMR-reader, maar dat is niet in alle gevallen of op elk OS hetzelfde. De melding zegt mij inhoudelijk overigens weinig en ik kan er [weinig over vinden](https://www.google.com/search?q="Will+not+install+to+the+user+site+because+it+will+lack+sys.path+precedence+to+pip+in").

Welke versie van je OS gebruik je? Je kunt dit zien met `lsb_release -a`.

Roel0412, 2021-06-07:
Dit is de OS versie:

[image]

dennissiemensma, 2021-06-08:
Dat is iig dezelfde OS-versie die ik ook gebruik. Ik kan het zelf niet reproduceren. Je zou kunnen proberen om de virtualenv voor DSMR-reader opnieuw aan te maken, zonder het `--system-site-packages` argument, alleen weet ik niet of dat het ook daadwerkelijk oplost. 
Als ik heel eerlijk ben zou ik het laten voor wat het is als DSMR-reader verder goed draait. Pip is geen onderdeel van DSMR-reader en opzich draai je al een relatief nieuwe Pip-versie van eerder dit jaar. Er is namelijk ook een kans dat het stuk gaat en mocht je zelf niet helemaal thuis zijn in Python of pip, dan kan het wat lastig zijn om te fixen. 
Daarnaast ben ik van plan om vanaf DSMR-reader 5.x "Poetry" te gaan gebruiken, wat het aanmaken van de virtualenv en pip zelf doet en het hopelijk nog wat makkelijker maakt ipv zelf dit soort dingen te moeten aanmaken.

Roel0412, 2021-06-09:
Ahh, dan laat ik het zo wel. Het werkt inderdaad, alleen update ik braaf die pip versie als ie dat aangaf.

Ik laat het dan wel zo, want zoveel verstand heb ik er inderdaad niet van :)

Bedankt in ieder geval voor je antwoord!

closed 2021-06-09 by dennissiemensma