dsmrreader/dsmr-reader #1406: dsmr_backend loopt vast na verliezen database connectie. Zou deze automatisch kunnen herstarten of wellicht een notificatie kunnen sturen?

vincentweeda, 2021-06-09:
Beste, 

Ik kwam er achter dat wanneer mijn database server niet bereikbaar is dat de gehele dmsr_backend crashed. Dat is op zich logisch want geen database.... maar nu vroeg ik mij af is het mogelijk om een notificatie te versturen bij meer dan x fouten? 
Of misschien na x bepaalde tijd dat de backend zichzelf probeert te herstellen. 

Als ik doormiddel van het commando "supervisorctl restart dmsr_backend" de service herstart dan loopt alles weer netjes door trouwens. 

```
OS: Raspbian 
Software versie: DMSR/v4.16.3
Posgresql: Versie 12 (Redhat Linux 8)
```

```
dsmr_backend                     RUNNING   pid 4671, uptime 0:07:23
dsmr_datalogger                  RUNNING   pid 13964, uptime 5 days, 14:57:58
dsmr_webinterface                RUNNING   pid 7021, uptime 14 days, 10:21:49
```

```

DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             241762
    Consumption records electricity / gas (est.)                                                  40904 / 950

UNRESOLVED ISSUES
    Too many unprocessed readings: 2599                                                           7 hours ago
    Day statistics are lagging behind                                                      1 day, 8 hours ago

```

**Laatste 200 regels uit het logbestand**

```
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 130, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: could not connect to server: No route to host
	Is the server running on host "mijndatabaseserver" and accepting
	TCP/IP connections on port 5432?

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 46, in handle
    self.data = self.initialize()
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 22, in initialize
    self.sleep_time = BackendSettings.get_solo().process_sleep
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/solo/models.py", line 55, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 573, in get_or_create
    return self.get(**kwargs), False
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 425, in get
    num = len(clone)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1308, in _fetch_all
    self._result_cache = list(self._iterable_class(self))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1154, in execute_sql
    cursor = self.connection.cursor()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 235, in _cursor
    self.ensure_connection()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 130, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
django.db.utils.OperationalError: could not connect to server: No route to host
	Is the server running on host "mijndatabaseserver" and accepting
	TCP/IP connections on port 5432?

Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 130, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: could not connect to server: No route to host
	Is the server running on host "mijndatabaseserver" and accepting
	TCP/IP connections on port 5432?

....enzovoorts, melding herhaald zich tot dat ik de backend herstart
```

dennissiemensma, 2021-06-09:
Bedankt voor je melding! Weet je zeker dat de backend en datalogger niet automatisch herstarten bij het wegvallen van de database? 

Want als ik het zelf even probeer door simpelweg PostgreSQL te stoppen, dan lijkt dat namelijk wel te gebeuren. Want ze zouden zichzelf moeten afsluiten waardoor Supervisor ze weer start en ze het opnieuw proberen. Vervolgens zou die cirkel rond moeten blijven gaan. 
Alleen de webinterface herstart zichzelf niet, maar detecteert wel als verbinding terug is (lijkt het). Als ik de DB-verbinding herstel en refresh in de browser dan pakt die hem weer op. Al zijn de datalogger en backend het belangrijkste.

dennissiemensma, 2021-06-09:
En qua notificaties is het wat lastig, want de instellingen waar een notificatie heen moet staan doorgaans in de database. Los van in hoeverre zoiets in DSMR-reader hoort. 
Alles is uiteraard te maken, maar wellicht is het veel makkelijker op te lossen met een (gratis) tool die van alles kan monitoren, waaronder DSMR-reader zelf (want als DSMR-reader niet draait komen er sowieso geen notificaties.

vincentweeda, 2021-06-10:
> Bedankt voor je melding! Weet je zeker dat de backend en datalogger niet automatisch herstarten bij het wegvallen van de database?
>
> Want als ik het zelf even probeer door simpelweg PostgreSQL te [...]

De reden dat ik denk dat hij niet herstart is het aantal blijft oplopen bij " Too many unprocessed readings:" tot dat ik zelf de backend een schop geef. 
Op dat moment heb ik al wel zaken gechecked als de beschikbaarheid van de database server. 

Het gebeurt eens per week (waarom dat weet ik nog niet) maar ik kan eventueel de logging wat hoger zetten met de hoop dat ik iets waardevols aantref. 

> En qua notificaties is het wat lastig, want de instellingen waar een notificatie heen moet staan doorgaans in de database. 

Zo vermoeden had ik al, dat kan ik dan oplossen via een tool als monit of iets dergelijks

dennissiemensma, 2021-06-20:
@vincentweeda heb je hier nog aanvullingen op?

Overigens qua monitoring zou je nog deze comment (en vooral alles daaronder) kunnen checken: https://github.com/dsmrreader/dsmr-reader/issues/1086#issuecomment-693531942
Het gaat welliswaar over Nagios, maar ik verwacht dat een soortgelijke check voor iets als Monit ook vrij eenvoudig in elkaar te zetten is.

vincentweeda, 2021-06-21:
Helaas niet, moet zeggen dat hij stabiel loopt sinds de melding. 

Denk niet dat het enige waarde heeft om dit topic open te laten staan.  Ik heb nog steeds de logging hoger staan, voor het geval dat. 

Mocht er echt iets uitkomen waar je wat aan kan hebben dan zal ik mij weer melden.

dennissiemensma, 2021-06-22:
Bedankt voor de aanvulling. Laat het in dat geval idd maar weten als het ooit alsnog weer speelt.

Je kunt overigens overwegen om de debug-logging weer uit te zetten, mits je het nu op een SD-kaartje draait. Dat scheelt aanzienlijk wat writes, al weet ik niet 100% zeker of dat wel of geen invloed heeft op de levensduur van de SD-kaartjes.

closed 2021-06-22 by dennissiemensma