dsmrreader/dsmr-reader #1408: crc errors

jhammega, 2021-06-10:
Hi Dennis,

We went to new home therefor I performed a clean install of DSMR-reader v4.16.3 on Raspberry Pi 3 Model B Plus Rev 1.3.
Went through step-by-step install. All went well. I have a V5 meter and use a regular P1 cable.
Cable looks in good working condition, not attached to any powerlines etc.

Output of command "cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q" provides output:
Connected.

!8BCB
/XMX5LGF001045644600

1-3:0.2.8(50)
0-0:1.0.0(210610104302S)
0-0:96.1.1(453030363'303035363434363035303230)
1-0:1.8.1(000095.052*kWh)
1-0:1.8.2(000038.833*kWh)
1-0:2.8.1(000126.176*kWh)
1-0:2.8.2(000283.360*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.677*kW)
0-0:96.7.21(00011)
0-0:96.7.9(00002)
1-0:99.97.0(2)(0-0:96.7.19)( 00101010000W)(0000001050*s)(00010101000W)(0000071009*s)
1-0:32.32.0(00009)
1-0:52.32.0(00009)
1-0:72.32.0(00008)
1,0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00167)
0-0:96.13.0()
1-0:32.7.0(246.0*V)
1-0:52.7.0(244.5*V)
1-0:72.7.0(251.3*V)
1-0:31.7.0(001*A)	
                        1-0:51.7.0(002*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.202*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.000*kW)
1-0222.7.0(00.000*kW)
1-0:42.7.0(00.433*kW)
1-0:62.7.0(00.445*kW)
!C19D
/XMX5LGF0010456446050

1-3:0.2.8(50)
0-0:.0.0(210610104303S)
0-0:96.1.1(4530303637303035363434363035303230)
1-0:1.8.1(000095.052*kGh)
1-0:1.8.2(000038.833*kWh)
1-0:2.8.1(000!26.176*kWh)
1-0:2.8.2(000283.360*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.689*+W)
0-0:96.7.21(00011)
0-0:96.7.9(00002)
1-0:99.97.0(2)(0-0:96.7.19)(000101010000V)(0000001050*s)(0001101000W)(0000071009*s)
1-0:32.32.0(00009)
1-0:52.32.0(00009)
1-0:72.32.0(00008)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00167)
0-0:96.13.0()
1-0:32.7.0(246.0*V)
1-0:52.7.0(244.5*V)
1-0:72.7.0(251.3*V)
1-0:31.7.0(001*A)
1-0:51.7.0(002*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.194*kW)
1-0:41&7.0(00.000*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.435*kW)
1-0:62.7.0(00.448*kW)
!7641
/XMX5LGcu: Got hangup signal

Disconnected.

I do not see any comma's or other invalid characters. I see a few spaces in line 25.
I did check all "issues" regarding crc errors for dsmr reader on Github.
Did not try disabling crc yet.

I disabled "gas" meter reading as we not have gas in our new house.
We have "warmtepomp" and 18 350wp "solar panels".

After enabling debug in ".env" and restart all via supervisor. I see in datalogger file:

2021-06-10 10:26:15,684 DEBUG    datalogger   telegram_to_reading              74 | Received telegram:
/XMX5LGF0010456446050

1-3:0.2.8(50)
0-0:1.0.0(210610102613S)
0-0:96.1.1(4530303627303035363434363035303230)
1-0:!.8.1(000095.052*kWh!
1-0:1.8.2(000038.828*kWh)
1-0:2.8.1(000126.176*kWh)
1-0:2.8.2(000283.118*kWh)
0-0:96.14.0(0002)
1-021.7.0(00.000*kW)
1-0:2.7.0(00.653*kW)
0-0:96.7.21(00011)
0-0:96.7.9(00002)
1-0:99.97.0(2)(0-0:96.7.19)(000101010000W)(0000001050*s)(000101010000W)(0000071009*s)
1-0:32.32.0(00009!
2021-06-10 10:26:15,695 WARNING  datalogger   telegram_to_reading              81 | Rejected telegram: Failed to perform CRC validation because the telegram is incomplete. The checksum and/or content values are missing.
2021-06-10 10:26:15,705 DEBUG    mixins       run_loop                         73 | dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
2021-06-10 10:26:20,720 INFO     dsmr_datalogger_api_client read_telegram                    30 | [2021-06-10 10:26:20.720015] Opening connection "/dev/ttyUSB0" using options: {'baudrate': 115200, 'bytesize': 8, 'parity': 'N', 'stopbits': 1, 'xonxoff': 1, 'rtscts': 0}
2021-06-10 10:26:21,058 DEBUG    dsmr_datalogger_api_client read_telegram                    51 | [2021-06-10 10:26:21.058402] Read 100 Byte(s)
2021-06-10 10:26:21,392 DEBUG    dsmr_datalogger_api_client read_telegram                    51 | [2021-06-10 10:26:21.392682] Read 25 Byte(s)
2021-06-10 10:26:21,727 DEBUG    dsmr_datalogger_api_client read_telegram                    51 | [2021-06-10 10:26:21.727129] Read 685 Byte(s)
2021-06-10 10:26:22,061 DEBUG    dsmr_datalogger_api_client read_telegram                    51 | [2021-06-10 10:26:22.061652] Read 99 Byte(s)
2021-06-10 10:26:22,068 DEBUG    datalogger   telegram_to_reading              74 | Received telegram:
/XMX5LGF0010456446050

1-3:0.2.8(50)
0-0:1.0.0(210610102619S)
0-0:96.1.1(4530303637303035363434363035303230)
1-0:1.8.2(000038.828*kWh)
1-0:2.8.1(000126.176*kWh)
1-0:2.8.2(000283.119*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.00*kW)
1-0:2.7.0(00.646*kW)
0-0:)6.7.21(00011)
0-0:96.7.9(00002)
1-0:99.)7.0(2)(0-0:96.7.19)(000101010000W)(000001050*s)(00001010000W)(0000071009*s)
1-0:32.32.0(00009)
1-0:52.32.0(00009)
-0:72.32.0(00008)
1-0:32.26.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00167)
0-0:96.13.0()
1-0:32.7.0(246.5*V)
1-0:52.7.0(244.4*V	
1-0:72.7.0(251.1*V)
1-0:31.7.0(001*A)
1-0:51.7.0(001*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00,000*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.106*kW)
1-0:42.7.0(00.217*kW)
1-0:62.7.0(00.323*kW)
!0AA2
2021-06-10 10:26:22,085 WARNING  datalogger   telegram_to_reading              81 | Rejected telegram: Invalid telegram CRC. The calculated checksum '42350' (A56E) does not match the telegram checksum '2722' (0AA2)
2021-06-10 10:26:22,090 DEBUG    mixins       run_loop                         73 | dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
2021-06-10 10:26:27,105 INFO     dsmr_datalogger_api_client read_telegram                    30 | [2021-06-10 10:26:27.105481] Opening connection "/dev/ttyUSB0" using options: {'baudrate': 115200, 'bytesize': 8, 'parity': 'N', 'stopbits': 1, 'xonxoff': 1, 'rtscts': 0}
2021-06-10 10:26:27,443 DEBUG    dsmr_datalogger_api_client read_telegram                    51 | [2021-06-10 10:26:27.443664] Read 195 Byte(s)
2021-06-10 10:26:27,777 DEBUG    dsmr_datalogger_api_client read_telegram                    51 | [2021-06-10 10:26:27.777650] Read 609 Byte(s)
2021-06-10 10:26:28,111 DEBUG    dsmr_datalogger_api_client read_telegram                    51 | [2021-06-10 10:26:28.111827] Read 7 Byte(s)
2021-06-10 10:26:28,124 DEBUG    datalogger   telegram_to_reading              74 | Received telegram:
/XMX5LGF0010456446050

1-3:0.2.8(50)
0-0:1.0.0(21061002625S)
0-0:96.1.1(4530303637303035363434163035303230)
1-0:1.8.1(000095.052*kWh)
1-0:1.8.2(000038.828*kWh)
1-0:2.8.1 000126.176*kWh)
1,0:2.8.2(000283.120*kWh)
0-0:96.14.0(0002)
1-0:1.7*0(00.000*kW)
1-0:2.7.0(00,631*kW)
0-0:96.7.21(00011)
0-0:96.7.9(00002)
1-0:99.97.0(2)(0-0:96.7.19)(000101010000W)(0000001050*s)(000101010000W)(0000071009*s)
1-0:32.32.0(00009)
1-0:52.32.0(00009)
1-0:62.32.0(00008)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00167)
0-0:96.13.0()
1-0:32.7.0(246.6*V)
1-0:52.7.0(244.4*V)
1-0:72.7.0(251.4*V)
1-0:31.7.0(001*A)
1-0:51.7.0(001*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.109*kW)
1-0:42.7.0(00.199*kW)
1-0:62.7.0(00.323*kW)
!31B4
2021-06-10 10:26:28,159 WARNING  datalogger   telegram_to_reading              81 | Rejected telegram: Invalid telegram CRC. The calculated checksum '52167' (CBC7) does not match the telegram checksum '12724' (31B4)
2021-06-10 10:26:28,170 DEBUG    mixins       run_loop                         73 | dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s

So in this output there is a comma. However in the output from "cu" there does not look to be any errors.
Should I invest in a new P1 cable?

Thanks. I appreciate your help and ofcourse your great software!

jhammega, 2021-06-10:
DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                                  -
    Consumption records electricity / gas (est.)                                                        - / -

UNRESOLVED ISSUES
    No day statistics found                                                                               now

closed 2021-06-10 by jhammega

dennissiemensma, 2022-11-01:
@jhammega in favor of #1733, did the new cable solve your issue? Also, is it specific for the E360 meter?

jhammega, 2022-11-01:
> @jhammega in favor of #1733, did the new cable solve your issue? Also, is it specific for the E360 meter?

@dennissiemensma. I ordered a new cable first and that did NOT solve the issue. Then I did conclude it had to be the E360 meter, cause in previous house I did not have any issue with the same cable. So I called LIANDER and asked to replace the E360 meter. I now have a ISKRA SMR 5.0 AM550-TD2 GPRS meter and no issues at all. I even connected a Smart Gateways Slimme meter WiFi gateway.

dennissiemensma, 2022-11-01:
Oh that is some great info, thanks for sharing and responding so quick. I'll pass it on!