dsmrreader/dsmr-reader #1412: ik loop vast bij ehe

scannie67, 2021-06-19:
> 🇳🇱 Lees eerst [deze info](https://github.com/dsmrreader/dsmr-reader/blob/v4/CONTRIBUTING.md) om jezelf te helpen.
> 
> 🇬🇧 First read [this info](https://github.com/dsmrreader/dsmr-reader/blob/v4/CONTRIBUTING.md) for self-service.

scannie67, 2021-06-19:
Op dit punt krijg ik een foutmelding tijdens de installatie:
sudo cp /home/dsmr/dsmr-reader/dsmrreader/provisioning/supervisor/dsmr_datalogger.conf /etc/supervisor/conf.d/

de foutmelding:
cp: cannot stat '/home/dsmr/dsmr-reader/dsmrreader/provisioning/supervisor/dsmr_datalogger.conf': No such file or directory

Iemand tips waar het mis gaat?

dennissiemensma, 2021-06-19:
Bedankt voor je vraag. In een van de eerdere stappen heb je als het goed is een kopie van dsmr-reader in /home/dsmr/dsmr-reader/ gezet. Bestaat die map wel?

closed 2021-06-28 by dennissiemensma