dsmrreader/dsmr-reader #1413: Energiecontracten herberekenen

vincedh, 2021-06-28:
Hoi Dennis,
M'n 'meterkast-pi' (die doorstuurt naar DSMR-reader op m'n nuc) en DSMR zelf is een half dagje offline geweest. Hierdoor kloppen m'n waardes in het Energiecontracten scherm niet meer (teruggeleverd is anders dan dat ik zie in het rapport van de energieleverancier wat zij hebben uitgelezen). 

Is er een manier om deze waardes opnieuw te laten berekenen, op basis van dagtotalen oid?

vincedh, 2021-06-28:
Hartelijk dank weer voor je snelle antwoord, ga ik dat doen!

closed 2021-06-28 by vincedh

dennissiemensma, 2021-06-29:
@vincedh heeft dit het uiteindelijk ook opgelost?

vincedh, 2021-06-30:
Hoi Dennis,
Ja helemaal gelukt, tnx!

rkollman, 2021-07-03:
Gewoon een idee. Het zou leuk zijn dat je verschillende energiecontracten invult en die dan kunt koppelen aan je voorgaande verbruik. Zo kan je dan zelf zien wat nu eigenlijk het beste voor jou situatie is om een andere energieleverancier te kiezen.

Staat zoiets toevallig al op de roadmap of vraag ik iets onhaalbaars?

dennissiemensma, 2021-07-03:
@rkollman bedankt voor je suggestie! Ik denk dat je een leuk idee hebt, alleen is dat voor een groot deel al mogelijk via commerciele vergelijkingssites zoals gaslicht/independer/etc. 

Daarnaast zijn de (kale) tarieven slechts een deel van de kosten, maar om een goede beslissing te nemen qua leverancier zijn er ook nog andere factoren zoals contractduur, vaste kosten en eventuele kortingen. Dit zijn allemaal zaken die buiten DSMR-reader vallen, maar wel belangrijk kunnen bij je beslissing. Dus ik denk dat je uiteindelijk beter af bent bij zo'n vergelijkingssite.