dsmrreader/dsmr-reader #1414: België: data uitlezen lukt niet: Too many unprocessed readings

brvdboss, 2021-07-03:
Hallo,

ik krijg steeds de fout dat de data telegrammen niet verwerkt kunnen worden.  Belgische meter, 3 fasen en geen gas. (ook volledig nachttarrief meter, maar die is niet verbonden enkel de dag)

Het datagram ziet er als volgt uit:

```
/FLU5\253769484_A

0-0:96.1.4(50215)
0-0:96.1.1(3153414733313030303932383637)
0-0:1.0.0(210703145113S)
1-0:1.8.1(002158.457*kWh)
1-0:1.8.2(003549.240*kWh)
1-0:2.8.1(006144.366*kWh)
1-0:2.8.2(002495.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(04.331*kW)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(02.133*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(02.197*kW)
1-0:32.7.0(234.6*V)
1-0:52.7.0(000.0*V)
1-0:72.7.0(235.0*V)
1-0:31.7.0(009.19*A)
1-0:51.7.0(015.87*A)
1-0:71.7.0(009.39*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
!F0F7
```

debug info dump:

```

DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.9.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                             "None" / "101"

DATA
    Telegrams total (est.)                                                                                259
    Consumption records electricity / gas (est.)                                                        - / -

UNRESOLVED ISSUES
    Too many unprocessed readings: 289                                                       an hour from now
    No day statistics found                                                                               now

```

Ik zie op zich niets vreemd in de logs en het publiceren via json (volledig telegram) lijkt gewoon te lukken.

```
2021-07-03 15:09:55,259 DEBUG    schedule     execute_scheduled_processes      28 | SP: 0 backend service(s) ready to run
2021-07-03 15:09:55,277 DEBUG    persistent_clients run                              30 | CLIENTS: Running 1 active client(s)
2021-07-03 15:09:55,289 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-07-03 15:09:56,298 DEBUG    schedule     execute_scheduled_processes      28 | SP: 0 backend service(s) ready to run
2021-07-03 15:09:56,315 DEBUG    persistent_clients run                              30 | CLIENTS: Running 1 active client(s)
2021-07-03 15:09:56,326 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-07-03 15:09:57,333 DEBUG    schedule     execute_scheduled_processes      28 | SP: 0 backend service(s) ready to run
2021-07-03 15:09:57,349 DEBUG    persistent_clients run                              30 | CLIENTS: Running 1 active client(s)
2021-07-03 15:09:57,353 DEBUG    broker       run                              75 | MQTT: Processing 1 message(s)
2021-07-03 15:09:57,354 DEBUG    broker       run                              78 | MQTT: Publishing queued message (#407) for dsmr/json: {"id": 407, "timestamp": "2021-07-03T15:09:57.260Z", "electricity_delivered_1": "2158.457", "electricity_returned_1": "6144.366", "electricity_delivered_2": "3549.240", "electricity_returned_2": "2496.226", "electricity_currently_delivered": "0.000", "electricity_currently_returned": "3.322", "phase_currently_delivered_l1": "0.000", "phase_currently_delivered_l2": "0.000", "phase_currently_delivered_l3": "0.000", "extra_device_timestamp": null, "extra_device_delivered": null, "phase_currently_returned_l1": "1.832", "phase_currently_returned_l2": "0.000", "phase_currently_returned_l3": "1.489", "phase_voltage_l1": "234.2", "phase_voltage_l2": "0.0", "phase_voltage_l3": "234.2", "phase_power_current_l1": 7, "phase_power_current_l2": 12, "phase_power_current_l3": 6}
2021-07-03 15:09:57,354 DEBUG    broker       on_log                          158 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m125), 'b'dsmr/json'', ... (773 bytes)
2021-07-03 15:09:57,356 DEBUG    broker       run                              97 | MQTT: Waiting for message (#407) to be marked published by broker
2021-07-03 15:09:57,356 DEBUG    broker       on_log                          158 | MQTT: (Paho on_log) Received PUBREC (Mid: 125)
2021-07-03 15:09:57,357 DEBUG    broker       on_log                          158 | MQTT: (Paho on_log) Sending PUBREL (Mid: 125)
2021-07-03 15:09:57,357 DEBUG    broker       run                              97 | MQTT: Waiting for message (#407) to be marked published by broker
2021-07-03 15:09:57,359 DEBUG    broker       on_log                          158 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 125)
2021-07-03 15:09:57,360 DEBUG    broker       run                             105 | MQTT: Deleting published message (#407) from queue
2021-07-03 15:09:57,374 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-07-03 15:09:58,383 DEBUG    schedule     execute_scheduled_processes      28 | SP: 0 backend service(s) ready to run
2021-07-03 15:09:58,402 DEBUG    persistent_clients run                              30 | CLIENTS: Running 1 active client(s)
2021-07-03 15:09:58,412 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-07-03 15:09:59,421 DEBUG    schedule     execute_scheduled_processes      28 | SP: 0 backend service(s) ready to run
2021-07-03 15:09:59,438 DEBUG    persistent_clients run                              30 | CLIENTS: Running 1 active client(s)
2021-07-03 15:09:59,448 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
```

Enig idee?

Ik heb zowel Nederland als België in de instellingen geprobeerd met gelijkaardig resultaat.

De installatie is de docker image van xirixiz

Alvast bedankt!

dennissiemensma, 2021-07-03:
Bedankt voor je melding. Ik ben nu niet in de gelegenheid om te debuggen voor je, maar kan het te maken hebben met de systeemtijd of die van de DB? Want ik zie de debug info met issues aangeven "An hour from now" terwijl dat doorgaans iets is als "An hour ago" (als ik het me goed herinner). Dus dattie onterecht denkt dat het telegram nog in de toekomst ligt en daardoor (nog) niet verwerkt. Het is een vermoeden hoor, meer niet. Zoek vooral ook even in oude issues hier mbt Docker en tijdzones.

brvdboss, 2021-07-03:
Je bent een held!

de entries in de DB stonden inderdaad 2u in de toekomst
dit aangepast door aanpassing in de docker config en nu werkt het wel!

Bedankt!

closed 2021-07-03 by brvdboss