dsmrreader/dsmr-reader #1415: data wordt niet meer bijgehouden

sjaakveenendaal, 2021-07-05:
```

DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 168h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             101733
    Consumption records electricity / gas (est.)                                                18673 / 10671

UNRESOLVED ISSUES
    Geen recente metingen ontvangen                                                             3 uur geleden

```

dennissiemensma, 2021-07-05:
Bedankt voor je melding. Ik mis wat je eventueel al zelf hebt geprobeerd. 

Je kunt het beste kijken of je datalogger nog draait. Eventueel deze herstarten. En anders de logging van de datalogger bekijken: https://dsmr-reader.readthedocs.io/en/v4/how-to/troubleshooting/logfiles.html
Als je daar nog niets in kan vinden, debug-logging inschakelen (staat op dezelfde pagina).

sjaakveenendaal, 2021-07-05:
Heb wat dingen geprobeerd maar werkt nog steeds niet

 

The programs included with the Debian GNU/Linux system are free software;

the exact distribution terms for each program are described in the

individual files in /usr/share/doc/*/copyright.

 

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent

permitted by applicable law.

Last login: Mon Jul  5 22:34:01 2021 from 192.168.178.36

***@***.***:~ $ sudo supervisorctl status

dsmr_backend                     RUNNING   pid 20713, uptime 0:11:00

dsmr_datalogger                  STARTING

dsmr_webinterface                RUNNING   pid 20711, uptime 0:11:00

***@***.***:~ $ sudo supervisorctl restart dsmr_datalogger

dsmr_datalogger: stopped

dsmr_datalogger: started

***@***.***:~ $ sudo supervisorctl restart dsmr_backend

dsmr_backend: stopped

dsmr_backend: started

***@***.***:~ $

dennissiemensma, 2021-07-05:
Het lijkt erop dat de datalogger telkens uit zichzelf stopt en start, gezien de andere processen al 11 minuten draaiden. Je kunt het beste in de log file van de datalogger kijken, zie bovenstaande documentatie voor meer details.

sjaakveenendaal, 2021-07-06:
Welk document?

dennissiemensma, 2021-07-06:
In mijn eerdere comment https://github.com/dsmrreader/dsmr-reader/issues/1415#issuecomment-874302038

sjaakveenendaal, 2021-07-07:
Hoi Dennis ik krijg het niet aan de gang zo jammer jullie hebben zo een mooi programma gemaakt maar is voor iemand die niet veel kennis heeft niet te doen ik denk dat ik maar weer opnieuw ga installeren of heb jij nog een tip?

dennissiemensma, 2021-07-07:
Ik denk dat herinstalleren niet helpt omdat het aan de datalogger lijkt te werken. Zoals gezegd kun je het beste in de logfiles kijken, dat kan met:

```
sudo tail -f /var/log/supervisor/dsmr_webinterface*
sudo tail -f /var/log/supervisor/dsmr_datalogger*
sudo tail -f /var/log/supervisor/dsmr_backend*
```

Ik denk dat je de middelste nodig hebt. Wellicht zie je dan een foutmelding of oorzaak.

sjaakveenendaal, 2021-07-07:
Dan krijg ik deze melding

 

sudo tail -f /var/log/supervisor/dsmr_datalogger*

==> /var/log/supervisor/dsmr_datalogger.log <==

Current logging level set to "ERROR". More information can be found here: https:                 //dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-log                 ging.html

Current logging level set to "ERROR". More information can be found here: https:                 //dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-log                 ging.html

Current logging level set to "ERROR". More information can be found here: https:                 //dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-log                 ging.html

Current logging level set to "ERROR". More information can be found here: https:                 //dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-log                 ging.html

Current logging level set to "ERROR". More information can be found here: https:                 //dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-log                 ging.html

Current logging level set to "ERROR". More information can be found here: https:                 //dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-log                 ging.html

Current logging level set to "ERROR". More information can be found here: https:                 //dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-log                 ging.html

Current logging level set to "ERROR". More information can be found here: https:                 //dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-log                 ging.html

Current logging level set to "ERROR". More information can be found here: https:                 //dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-log                 ging.html

Current logging level set to "ERROR". More information can be found here: https:                 //dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-log                 ging.html

 

==> /var/log/supervisor/dsmr_datalogger-stderr---supervisor-QaeTzi.log <==

 

2021-07-07 20:05:11,370 ERROR    mixins       run_once                        10                 1 | dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. T                 raceback (most recent call last):

  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once

    self.run(data=self.data, **options)

  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogg                 er.py", line 29, in run

    telegram = next(self.telegram_generator)

  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_clien                 t.py", line 46, in read_telegram

    len(buffer)

RuntimeError: It took too long to detect a telegram. Check connection params. By                 tes currently in buffer: 0

 

 

==> /var/log/supervisor/dsmr_datalogger.log <==

Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html

 

==> /var/log/supervisor/dsmr_datalogger-stderr---supervisor-QaeTzi.log <==

2021-07-07 20:05:42,705 ERROR    mixins       run_once                        101 | dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):

  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once

    self.run(data=self.data, **options)

  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run

    telegram = next(self.telegram_generator)

  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 46, in read_telegram

    len(buffer)

RuntimeError: It took too long to detect a telegram. Check connection params. Bytes currently in buffer: 0

 

 

==> /var/log/supervisor/dsmr_datalogger.log <==

Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html

 

==> /var/log/supervisor/dsmr_datalogger-stderr---supervisor-QaeTzi.log <==

2021-07-07 20:06:14,096 ERROR    mixins       run_once                        101 | dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):

  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once

    self.run(data=self.data, **options)

  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run

    telegram = next(self.telegram_generator)

  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 46, in read_telegram

    len(buffer)

RuntimeError: It took too long to detect a telegram. Check connection params. Bytes currently in buffer: 0

 

 

==> /var/log/supervisor/dsmr_datalogger.log <==

Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html

dennissiemensma, 2021-07-07:
De melding is:
```
It took too long to detect a telegram. Check connection params
```

Er vanuitgaande dat je niets aan de instellingen of meter of kabel hebt verandert, kun je het eens met de hand proberen te meten:

```
sudo supervisorctl stop dsmr_datalogger
```

```
sudo su - dsmr
cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q
```
Als je nu max 10 seconden wacht, zou je iets van een telegram van je meter moeten zien.

Zo niet dan kan het zo zijn dat er iets met de kabel of slimme meter is. Alleen heb ik dat zelden eerder gezien. Problemen met de meter of kabel zijn er doorgaans direct al en niet na verloop van tijd of opeens.

Als je klaar bent met zelf proberen.
- Typ je `q.` + Enter om te stoppen met meten.
- Daarna:
```
logout
sudo supervisorctl start dsmr_datalogger
```

sjaakveenendaal, 2021-07-07:
Ik krijg geen data ook even een ander kabeltje geplaatst tussen pi en slimme meter maar helpt ook niet

 

 

***@***.***:~ $ sudo su - dsmr

(dsmrreader) ***@***.***:~/dsmr-reader $ cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q

Connected.

dennissiemensma, 2021-07-07:
Je zou nog kunnen proberen om een keer op een geschikt moment de slimme meter even van de stroom af te halen en daarna weer erop te zetten. Ik weet niet of dat uberhaupt iets doet bij de meter of iets oplost, maar wie weet. Zoals gezegd heb ik dit niet eerder gezien.

Als dat niet helpt dan kun je wellicht even een keer bellen naar je netbeheerder om te vragen of zij je slimme nog wel kunnen uitlezen of dat ze nog andere tips hebben. Helemaal omdat je dus zelf ook al met een andere kabel hebt geprobeerd.

dennissiemensma, 2021-07-07:
En je kunt uiteraard nog je Pi herstarten, maar ik neem aan dat je dat ook al geprobeerd hebt.

sjaakveenendaal, 2021-07-07:
Ik heb er een splitter tussen zitten die is dus stuk ik heb hem ertussenuit gehaald en nu werkt het wel dank voor je hulp

closed 2021-07-07 by dennissiemensma