dsmrreader/dsmr-reader #1416: Backup wordt op verkeerde tijd weggeschreven

reboot1983, 2021-07-06:
Ik merk een vreemd dingetje op. De backup bestanden worden (zoals ik er naar kijk) om 02:00 weggeschreven i.p.v. middernacht, zoals dit is ingesteld.
Té toevallig is de offset 2 uur (tijdzone met zomertijd) en dus dacht ik dat ik het verkeerd interpreteerde, maar na zelf een bestandje te schrijven komt hier toch echt de juiste datum/tijd bij te staan.

Ook als ik de backup actie trigger vanuit de GUI staat hier de huidige tijd. Mijn eerste indruk is dat er iets in de backup sequence niet goed gaat en het lijkt erop dat ie geen rekening houdt met de timezone. 

[image]
[image]

dennissiemensma, 2021-07-06:
Bedankt voor je melding! Ik zie hier hetzelfde en vermoed ook dat het aan de tijdzone ligt. Mogelijk bij het [berekenen van het eerstvolgende moment](https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_backup/services/backup.py#L45-L53).

Ik zal er in een latere release naar kijken.

dennissiemensma, 2021-09-19:
Het lijkt er inderdaad op dat de tijdzone niet lekker doorkomt, al is het me een raadsel waarom.

```
SP: Rescheduled "Create daily backup" to 2021-09-20 06:00:00+00:00
```

Hoe dan ook zal ik forceren dat die de goed tijdzone pakt, voor de v4.18 release.

closed 2021-09-19 by dennissiemensma