dsmrreader/dsmr-reader #1417: Meerdere meters in de webinterface

skinkie, 2021-07-08:
Kun je op dit moment meerdere remote meters naar 1 dsmr-reader backend sturen, zodat je van verschillende panden informatie kan vergelijken?

dennissiemensma, 2021-07-08:
Bedankt voor je vraag. Dit is helaas niet op een gebruiksvriendelijke manier mogelijk voor je. Het project is ooit opgezet voor thuisgebruik en daarbij 1 slimme meter. Wat je wel kunt doen is ervoor kiezen om meerdere instanties te draaien via Docker. Alleen vereist dat wel wat technische kennis, maar dan kun je nog steeds niet de gegevens binnen 1 scherm of tabblad tonen.

closed 2021-07-08 by dennissiemensma

skinkie, 2021-07-08:
@dennissiemensma bedankt voor je reactie. Is de code voorbereid op meerder meters, of moet daarvoor echt alles om?

dennissiemensma, 2021-07-08:
Nee daar zit helemaal geen ondersteuning voor in. Ook het wisselen/vervangen van een meter is (eenmalig) een issue. En aan de visuele zijde is ook alles gefocused op 1 meter.

Wat je misschien nog wel kan doen is kijken of je je slimme meters kan laten loggen naar een timeseries database of iets en daar dan een visuele tool tegen laten praten. 
Als ik DSMR-reader tegenwoordig opnieuw zou bouwen, zou ik het gros in bijvoorbeeld een InfluxDB of ElasticSearch doen.