dsmrreader/dsmr-reader #1419: Edge on android not showing graphs

BertHankes, 2021-07-10:
When using Edge on Android all the charts give an error and are not shown.

All the .js files on /static/dsmr_frontend/js/dsmr-reader/live/*.js are not shown because of "chart_object**?**.resize();"

E.g. in electricity.js
This 
$(window).resize(function () {
    echarts_electricity_graph?.resize();
});
should be:
$(window).resize(function () {
    echarts_electricity_graph.resize();
});

dennissiemensma, 2021-07-11:
Bedank voor je melding. De syntax die gebruikt wordt is vrij generiek en wordt door de meeste browsers ondersteund:

- https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Optional_chaining#browser_compatibility (zie de tabel onderaan de pagina)

Kan het zijn dat je een oude versie van Edge (< 80) gebruikt? Als ik bovenstaande tabel mag geloven, zou er wel ondersteuning voor moeten zijn. 

Overigens heb ik geen mogelijkheden om het inhoudelijk te testen. De browser-ondersteuning is in DSMR-reader is momenteel vrij generiek met Firefox en Chrome. De meeste zaken zullen in andere browsers ook werken, maar ik verifieer dat dus niet structureel.

BertHankes, 2021-07-11:
Ik zoek het even uit vanaf hier... Ga er even goed voor zitten! Als ik het heb gevonden kom ik bij je terug. 
Edge mobile is, zo lijkt het, niet gebouwd op Chromium. (versie 42, voor wat het waard is). Edge op windows: flawless!

Prachtig programma overigens.. Net van 1 fase naar 3 fase gegaan (dus nieuwe meter) en naadloos door (wel even 2 archiefentries op 0 zetten)! Echt geweldig!

dennissiemensma, 2021-07-12:
Bedankt voor je aanvulling! Een hoop features in codetalen zijn er niet omdat ze noodzakelijk zijn, maar omdat ze het programmeren makkelijker of sneller maken. Deze en vele andere features zijn verbeteringen tov de "old skool" Javascript, waar DSMR-reader voor een groot deel is/was mee geschreven is.
In dit geval is het een alias voor checks want niet elke grafiek is altijd voor elke gebruiker beschikbaar. Overigens wordt het nog op veel meer plekken gebruikt in DSMR-reader en niet alleen bovenstaande.

Het is voor mij echter niet te doen om voor altijd "oude" code stijl te hanteren, dus op een gegeven moment zal het bijgewerkt worden.  Het is destijds juist omgezet naar deze variant, dus ik zal het zeker niet zomaar terugdraaien tenzij het voor een groot deel van de gebruikers of browsers breekt. 

Ik begrijp dat het wat lastig is voor specifiek jouw situatie, maar uiteindelijk ligt het aan de browser. Helemaal als alle andere browsers en platformen er al meer dan een jaar ondersteuning voor blijken te hebben. 
Het zou juist een reden moeten zijn om bij de ontwikkelaars van de mobile Edge aan te geven dat ze achter lopen, want anders krijgen we een tweede (mobiele) Internet Explorer,  qua uitzonderingen, waar niemand in de wereld op zit te wachten.

BertHankes, 2021-07-13:
Ik heb een scriptje dat na de update de zaak voor mij rechtzet. Dus geen probleem verder.. Omdat ik zelf ook programmeer heb ik verschillende browsers op telefoon.. vandaar dat ik deze tegenkwam.

Hoe dan ook.. Mooie software hoor!

closed 2021-07-13 by BertHankes