dsmrreader/dsmr-reader #1420: Feature request: unstacked phase

skinkie, 2021-07-10:
[image]

Gezien er nagenoeg geen apparaten zijn die gelijktijdig twee (of drie) fases gebruiken, is het wat betreft visualisatie (veel) handiger om een lijn grafiek te zien voor het gebruik per fase zonder dat deze is opgeteld. Je wilt immers kunnen zien welke fase wordt belast, want het cummulatieve gebruik is al af te lezen in recent electricity usage.

dennissiemensma, 2021-07-11:
Bedankt voor je suggestie! Het probleem is meer dat iedereen zijn eigen voorkeur heeft voor de eigen situaties. 

Wat voor jou handig is, werkt voor een andere gebruiker weer niet. Voor een deel kan ik dat oplossen met instelbare grafiekopties, maar ook dat houdt op een gegeven moment op.
Wat ik nog wel overweeg is om de bestaande checkbox voor wel/niet stapelen in het Archief door te trekken naar de Live grafieken, zodat dat ook daar te wisselen is. 

Of nog beter, nu ik er over nadenk, een manier aanbieden waarbij jullie de grafiekenconfig zelf kunnen overschrijven om te bereiken wat je wil. Dat betreft zowel grafiekstijl als stapelen en zelfs opties qua assen. Ik kan dan tzt wel assisteren met wat voorbeelden, want je bent absoluut niet de enige die grafieken net iets anders wil en waar het blijft zoeken naar een compromis.

dennissiemensma, 2022-07-10:
Ik zal voor de volgende release de "stack" optie ook toepassen op de fases en stroomsterktes bij Live grafieken.

Mocht dat toch teveel ongewenste effecten hebben bij andere gebruikers, dan zal ik het weer terugdraaien.

dennissiemensma, 2022-07-10:
## Beide varianten met fictieve/nepdata, grafieken 4 en 6

## Stacked
[image]

---

## Unstacked

[image]

closed 2022-07-10 by dennissiemensma