dsmrreader/dsmr-reader #1421: Feature request: line chart total consumed (laag+hoog)

skinkie, 2021-07-10:
Wat ik mis is een lijn grafiek welke de ouderwetse 'tellerstanden' plot over tijd. Een grafiek die van links onder naar rechtsboven loopt. Wat ik interessant zou vinden is als deze grafiek het laag en hoog tarief in dezelfde lijn zou zetten, maar de lijn (of achtergrond) van dit deel een andere kleur zou geven.

[image]

dennissiemensma, 2021-07-11:
Bedankt voor je suggestie. Opzich kan ik me vinden in je idee, alleen is het wel zo dat DSMR-reader standaard na verloop van tijd metingen verwijdert t.b.v. datagrootte. Uiteindelijk blijft er dan van elke dag slechts 1 meting over en is de nauwkeurigheid net zo hoog als die van de dagtotalen. 

Dat zeggende hebben kan ik me voorstellen dat de meterstanden sowieso wel handig zijn om in te zien. Alleen zullen dat dan eerder 4 of 5 lijnen zijn, namelijk voor elke teller. 
De manier die je hierboven vraagt is denk ik net iets te specifiek en gaat ook niet werken voor doordeweekse dagen icm met slechts 1 meting per dag. Alle tellers worden immers binnen dezelfde 24 uur bijgewerkt die geplot wordt.

Ik denk dat als je absolute vrijheid wilt qua renderen van je meterdata, je kunt overwegen het door te zetten naar een timeseries-database en dat vervolgens te plotten met een Grafana of soortgelijke tool. Er zijn ook meerdere gebruikers die dat linksom of rechtsom doen.

dennissiemensma, 2021-09-22:
Ik heb hier nogmaals naar gekeken en denk dat dit beter te realiseren is via een andere tool, eigenlijk het verhaal hierboven. Plus aanvulling in soortgelijk verzoek: https://github.com/dsmrreader/dsmr-reader/issues/1348#issuecomment-923339346

closed 2021-09-22 by dennissiemensma