dsmrreader/dsmr-reader #1422: Reset usage after meter swap

markisoke, 2021-07-13:
> 🇳🇱 Lees eerst [deze info](https://github.com/dsmrreader/dsmr-reader/blob/v4/CONTRIBUTING.md) om jezelf te helpen.
> 
> 🇬🇧 First read [this info](https://github.com/dsmrreader/dsmr-reader/blob/v4/CONTRIBUTING.md) for self-service.

I had my meter swapped today and now the usage is showing a negative number. Is there any way I can start counting again from the moment the meter has been swapped?

dennissiemensma, 2021-07-13:
Bedankt voor je vraag, je kunt hier meer informatie vinden: https://dsmr-reader.readthedocs.io/nl/v4/how-to/troubleshooting/smart-meter-replacement.html

markisoke, 2021-07-14:
> 
> 
> Bedankt voor je vraag, je kunt hier meer informatie vinden: https://dsmr-reader.readthedocs.io/nl/v4/how-to/troubleshooting/smart-meter-replacement.html

Bedankt Dennis, zojuist alles aangepast en ziet er nu weer goed uit.

dennissiemensma, 2021-07-14:
Top!

closed 2021-07-14 by dennissiemensma