dsmrreader/dsmr-reader #1423: Feature request: Auto backup using SSH/SCP using keys

reneposthumus, 2021-07-14:
It would be very welcome to be able to automatically backup all config and database to a NAS using SSH keys so that it doesn't have to be uploaded to dropbox since this IoT device doesn't have a connection to the internet.

Regards,
René

dennissiemensma, 2021-07-14:
Bedankt voor je verzoek. Ik denk eerlijk gezegd dat dit eerder iets is wat je op je Pi moet instellen, dan via een feature in DSMR-reader.

Afhankelijk van hoe bekend je bent met Linux en de commandline, zou je zoiets kunnen proberen:
- https://www.techrepublic.com/article/how-to-set-up-auto-rsync-backups-using-ssh/

In het `rsync` command wat je in de cronjob zet kun je dan de locatie van DSMR-reader opgeven als lokaal pad:
- `rsync -e 'ssh -p 22' -avzp /home/dsmr/dsmr-reader/ REMOTE_HOST_IP:/home/USER2/BACKUPS`

Als je de standaard-instellingen van DSMR-reader gebruikt, dan komen database-backups ook daar terecht (in `/home/dsmr/dsmr-reader/backups/`) en neemt die dat vanzelf mee. Vergeet het vooral niet even een keertje te testen, zoals ook in het artikel staat!

closed 2021-07-14 by dennissiemensma

reneposthumus, 2021-07-14:
Hi Dennis,

Dank voor je snelle reactie.

Ik ken meerdere softwarepakketten die dit wel ingebouwd hebben, maar niet gebaseerd op Rsync, maar SCP.
Als je in de interface een RSA keypair kunt genereren, de public key weergeven en IP adres en pad in kunt stellen, kun je daar volgens mij simpel een cronjob van maken.
Alleen het stukje configuratie op een NAS moet de gebruiker dan zelf even doen, maar dat is prima te overzien, daar heb ik wel wat docu voor gemaakt.

Met vriendelijke groet,
René Posthumus

dennissiemensma, 2021-07-14:
Bedankt voor je aanvulling! Alles is uiteraard te maken en ik begrijp je behoefte, maar realiseer je goed dat je iets specifiek voor jouw situatie vraagt. De kans is groot dat iemand anders er helemaal niets aan heeft en het juist via FTP wil, of het laten uploaden naar een backup-service via een API. 

Daarnaast weegt voor mij zwaar dat er al zoveel bestaande tools zijn, het zij `scp`, het zij `rsync`, die hetzelfde naadloos en snel kunnen doen. 
Bijvoorbeeld het uploaden naar Dropbox in DSMR-reader is een feature die er (achteraf gezien) dan ook niet thuis hoort en op lange termijn zal die dan ook uit DSMR-reader gaan, omdat er genoeg tools zijn die hetzelfde kunnen via de command line. Ook voor andere backup diensten. 

Daarom begin ik doorgaans niet aan zulke specifieke verzoeken, maar help ik de gebruikers wel op weg naar hoe ze het wel snel en zelf kunnen doen. 
Mijn bovenstaande suggestie staat daarom nog steeds en mocht je daar niet uit komen, dan wil ik uiteraard hier wel even met je mee kijken naar eventuele foutmeldingen in de verbinding.

reneposthumus, 2021-07-14:
Hi Dennis,

Ik snap het helemaal hoor, het is erg lastig om voor iedereen een passende oplossing voor backup te vinden en het in code te onderhouden. In mijn geval is het makkelijker om het op deze manier te doen omdat Home Assistant op dezelfde manier backups maakt.

Ik heb de oplossing al even snel in elkaar geschroefd, waarbij simpelweg dagelijks de hele map gekopieerd wordt naar een Synology NAS.

Hoe ik het geconfigureerd heb hoef ik je vast niet uit te leggen gezien je enorme kennis van Linux, maar dit is wat ik in het kort gedaan heb:

- user aangemaakt op de NAS, /homes/sshbackup/.ssh/authorized_keys gemaakt en rechten goed gezet
- keypair gegenereerd op de Pi en publieke deel naar de nas gezet
- scp command toegevoegd aan /etc/cron.daily/dsmr-reader: scp /home/dsmr/dsmr-reader/backups/*.* sshbackup@<nas-adres>:/volume1/Backup/DSMR/

Even afwachten of er morgen een backup staat, maar kan niet missen.

Nb. Is een backup van de database op deze manier voldoende om in geval van ellende, de hele configuratie en data terug te zetten?

Nogmaals dank voor je suggestie!

Met vriendelijke groet,
René Posthumus

dennissiemensma, 2021-07-14:
Je hoeft niet te wachten tot morgen om te testen. Je kunt het script gewoon uitvoeren en kijken of het werkt.

Check of je script hier bij staat:
```
run-parts --test /etc/cron.daily/
```

Zo ja, voer het uit met:
```
sh /etc/cron.daily/dsmr-reader
```
En kijk of het gelukt is.

Controleer ook even of alle submappen van die backup dir meekomen. En anders is `rsync` nog een alternatief.

Als dat niet werkt kun je ook nog `run-parts` uitvoeren zonder `--test`, alleen voert die dan wel **alle* scripts uit.

----

> Nb. Is een backup van de database op deze manier voldoende om in geval van ellende, de hele configuratie en data terug te zetten?

De backup bevat de meeste belangrijke informatie. Je zult een lege database moet aanmaken voor de restore, die vullen met de backup en daarna DSMR-reader herinstalleren.