dsmrreader/dsmr-reader #1424: Feature Request: add counters to CSV export/ archive view

Ferao, 2021-07-21:
> 🇳🇱 Lees eerst [deze info](https://github.com/dsmrreader/dsmr-reader/blob/v4/CONTRIBUTING.md) om jezelf te helpen.
> 
> 🇬🇧 First read [this info](https://github.com/dsmrreader/dsmr-reader/blob/v4/CONTRIBUTING.md) for self-service.

Is it possible to add the counter of the meter in the CSV export?
or/and see the counters in the archive -> data view section?

dennissiemensma, 2021-07-22:
Hi thanks for your suggestion. I'll add the counters to the CSV export, as it should be easy. 

And maybe in a later release to the Archive as well.

Ferao, 2021-07-23:
Perfect thanks

dennissiemensma, 2021-08-01:
Added for 4.17 release

closed 2021-08-01 by dennissiemensma

dennissiemensma, 2021-09-19:
Added in [v4.17](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.17.0)