dsmrreader/dsmr-reader #1426: DSMR lees probleem

H35am, 2021-07-30:
DSMR wilt niet meer lezen onder Home Assistant en spuugt steeds het zelfde error uit.

Logger: dsmr_parser.clients.protocol
Source: /usr/local/lib/python3.9/site-packages/dsmr_parser/clients/protocol.py:121
First occurred: 02:07:00 (1308 occurrences)
Last logged: 12:34:42

disconnected due to exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/serial_asyncio/__init__.py", line 114, in _read_ready
    data = self._serial.read(self._max_read_size)
  File "/usr/local/lib/python3.9/site-packages/serial/serialposix.py", line 595, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?) 

Probleem schijnt al langer aanwezig te zijn onder de gebruikers: 
https://community.home-assistant.io/t/dsmr-slimme-meter-problem/15956/4
https://community.home-assistant.io/t/dsmr-broken/255149

dennissiemensma, 2021-07-30:
Bedankt voor je melding! Hoewel DSMR-reader een deel van `dsmr_parser` gebruikt ([ander project, andere auteur](https://github.com/ndokter/dsmr_parser)), heeft DSMR-reader daar een [interne kopie](https://github.com/dsmrreader/dsmr-reader/tree/v4/dsmr_parser) van. 

Aan het pad te zien (`/usr/local/lib/python3.9/site-packages/dsmr_parser`) loop je tegen het issue aan van de stand-alone-versie van de parser. Je kunt het beste een issue maken in het betreffende project:
- https://github.com/ndokter/dsmr_parser

----

Desondanks ligt het vermoedelijk aan de manier van gebruiken of instellen, want de foutmelding lijkt simpelweg de constatering van de seriele poort lezer dat die er niet bij kan of niets ontvangt. 

Eerdere issues met dezelfde foutmelding binnen DSMR-reader:
- https://github.com/dsmrreader/dsmr-reader/issues/1058#issuecomment-664578430
- https://github.com/dsmrreader/dsmr-reader/issues/1033#issuecomment-647688050
- https://github.com/dsmrreader/dsmr-reader/issues/457#issue-307233125

Als ik zo vluchtig lees was het vaak een verkeerde versie van de verbinding DSMR 2 vs DSMR 4 meters bijvoorbeeld. Of meerdere applicaties die dezelfde poort uitlezen.

Hopelijk helpt je dit wat op weg.

H35am, 2021-07-30:
Dank voor het oppakken, het heeft altijd goed gewerkt en ineens is die stuk gegaan, was eerder ook onder RPI maar dan heel vaak, ben nu overgestapt naar Odroid met Hassio erop.

dennissiemensma, 2021-07-31:
Prima, fijn om te horen!

closed 2021-07-31 by dennissiemensma