dsmrreader/dsmr-reader #1427: Teruglevering niet zichtbaar in DSMR reader

Haasie, 2021-07-31:
Recent heb ik zonnepanelen laten plaatsen op mijn dak, helaas zijn de gegevens niet terug te vinden in DSMR reader.
het verbruik komt netjes binnen per telegram echter word deze neit geplot in de grafiek of waar dan ook.

ik gebruik dsmr reader icm docker en lees uit via P1->USB 
versie: 4-16-3 

dump: 

```
DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.9.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             188780
    Consumption records electricity / gas (est.)                                                 36329 / 3090
```

dennissiemensma, 2021-07-31:
Bedankt voor je vraag. Voor de duidelijkheid, welke gegevens zoek je exact? De teruglevering aan het net?

Zou je hier je telegram willen neerzetten? Als het kan zonder de serienummers van je meter(s), de lange cijferreeksen.

Haasie, 2021-07-31:
Hallo Dennis,

Bedankt voor je snelle reactie.

Het geen wat ik zoek is waarom mijn grafieken geen "teruglevering" aan het net laten zien.

```
1-0:1.8.1(008968.397*kWh)
1-0:2.8.1(000044.349*kWh)
1-0:1.8.2(011089.664*kWh)
1-0:2.8.2(000102.897*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(01.019*kW)
0-0:96.7.21(00002)
0-0:96.7.9(00002)
1-0:99.97.0(2)(0-0:96.7.19)(170325060157W)(0029754974*s)(160414214559S)(0037417488*s)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(005*A)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(01.019*kW)
0-1:24.1.0(003)
```
[image]
[image]

dennissiemensma, 2021-07-31:
Bednkt voor je aanvulling. 

- Check even of "teruglevering uitschakelen" niet hier aangevinkt staat, in tegenstelling van wat ik wel heb: `/admin/dsmr_backend/backendsettings/`
[image]

- En als dat goed staat, kijk of je wel teruglevering zie qua meterstanden in de metingen in DSMR-reader: `/admin/dsmr_datalogger/dsmrreading/`

Haasie, 2021-07-31:
Jeetje mina wat erg 😅🤭

Ik hem mij rot zitten zoeken, ook in de docs.

100X dank!

dennissiemensma, 2021-07-31:
Fijn om te horen!

closed 2021-07-31 by dennissiemensma