dsmrreader/dsmr-reader #1428: Hoog cpu gebruik

jimmyd-be, 2021-08-02:
Sinds kort is mijn Raspberry pi 4 extreem traag. Na wat onderzoek lijkt het erop dat de python en gunicorn van dsmr reader bijna 100% van de cpu opgebruikt.

Ik gebruik de remote datalogger en op de server zelf de laatste docker stack. De sever zelf is waar het allemaal traag loopt. De remote datalogger lijkt niets aan de hand.

Kan hier iets aan gedaan worden?

dennissiemensma, 2021-08-02:
Bedankt voor je melding. Dit kan verschillende oorzaken hebben, waarbij de meest voorkomende een te grote dataset is.

- Zou je een [debuginfo-dump willen delen hier](https://github.com/dsmrreader/dsmr-reader/blob/v4/CONTRIBUTING.md)?

dennissiemensma, 2021-08-02:
Voor de volledigheid, er zijn door de loop van de jaren verschillende issues geweest met load. Deels lag dat aan DSMR-reader zelf, deels aan een te trage CPU (vooral de oudere Pi-modellen destijds), deels lag dat aan I/O (de ene SD-kaart is niet de andere), deels lag dat aan een te hoge datatoevoer (lage sleeps) en in een enkel geval zelfs aan een mount die niet goed was en load veroorzaakte. 

Een algemene lijn aan meldingen en verbeteringen vind je hier terug: #774. Niet alles is de moeite waard om door te lezen, vooral vanaf de laatste paar comments denk ik: https://github.com/dsmrreader/dsmr-reader/issues/774#issuecomment-704203011

Hoe dan ook betekent dat niet dat dit ook voor jouw specifieke situatie geldt, maar bovenstaande is wel de algemene trend aan oorzaken en wat workarounds.

jimmyd-be, 2021-08-02:
Hier heb je mij mijn dump. In zal zeker eens kijken in #774  .

```

DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.9.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                             "None" / "101"

DATA
    Telegrams total (est.)                                                                            1031023
    Consumption records electricity / gas (est.)                                                42595 / 11653

```

dennissiemensma, 2021-08-02:
Bedankt voor je aanvulling. Het lijkt er op dat je een prima sleep + retentie gebruikt. Daar valt vaak de meeste winst te halen.

Wat nog wel opvalt is de *indicatie* dat er 1 miljoen metingen zijn. Gezien de twee relatief lage getallen eronder, ga ik er echter vanuit dat dat puur komt doordat de interne stats van PostgreSQL niet meer up te date zijn. DSMR-reader doet namelijk hier geen exacte count ivm de traagheid daarvan in PostgreSQL. 
Hoewel outdated stats doorgaans geen invloed op de performance heeft, kun je eventueel een handmatige opschoning/hertelling laten doen door PostgreSQL. Gezien je aangeeft Docker te gebruiken, probeer het tweede punt:
- https://github.com/xirixiz/dsmr-reader-docker#features

----

Mocht dat getal na de vacuum dat niet verkleinen, laat het even weten want wellicht werkt de retentie/opschoning dan niet helemaal lekker. Ter vergelijking, ik heb dit na 5+ jaar aan data (met een week retentie):
```

DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                    5.0s / 10.0s / 168h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             130464
    Consumption records electricity / gas (est.)                                               106935 / 41316

```

Dat performt bij mij op een Pi 4 tot nu toe prima. Maar zoals gezegd is de ene installatie/hardware de andere niet.

dennissiemensma, 2021-08-04:
@jimmyd-be ben je inmiddels nog wat wijzer geworden nav bovenstaande?

jimmyd-be, 2021-08-05:
Ik heb enkel nog maar het clean script gedraaid, verder nog geen tijd gehad om het verder te bekijken. Ik laat je iets weten als ik het heb kunnen bekijken

jimmyd-be, 2021-08-06:
Het draaien van het cleanup script heeft precies wel gewerkt. Bedankt voor de goede support en prachtig programma!

closed 2021-08-06 by jimmyd-be

dennissiemensma, 2021-08-06:
@jimmyd-be fijn om te horen!

Is dit aantal daarbij ook (flink) verminderd?

```
    Telegrams total (est.)                                                                            1031023
```

jimmyd-be, 2021-08-07:
Telegrams total staat momenteel op 1006650. De retintation staat bij mij wel op maand behouden.