dsmrreader/dsmr-reader #1429: Homey in combinatie met DSMR

Joriempie, 2021-08-05:
Hallo,

Onlangs overgestapt van een andere fabrikant naar DSMR. De handleiding voor de installatie was super duidelijk.

Nu probeer ik mijn Homey te verbinden met DSMR maar dit gaat niet lekker. Ik heb de handleiding doorgenomen zoals [hier](https://community.athom.com/t/p1-smartmeter-api/6150) stond onder het stukje: "Script for reading the data and push it to the api (DSMR-Reader)". Echter krijg ik nog steeds helaas geen waardes door op mijn Homey.

Excuus als ik hier niet aan het juiste adres ben.

dennissiemensma, 2021-08-05:
Bedankt voor je vraag. 

Heb je dit script in DSMR-reader gezet?
- https://gist.github.com/steffjenl/31bd083eeb9d0be04375b7695b9f2eaf

En vervolgens ook de plugin in je settings.py of envvar van DSMR-reader gezet?
- `DSMRREADER_PLUGINS=dsmr_plugins.modules.forward_raw_telegram_to_api`

En daarna DSMR-reader **herstart**?

Zo ja, check ook even of je foutmeldingen ziet in de `dsmr_backend` log:
- https://dsmr-reader.readthedocs.io/nl/v4/how-to/troubleshooting/logfiles.html

Joriempie, 2021-08-05:
Thanks voor de snelle reply!

Het stukje met "var" vind ik zelf erg onduidelijk. Maar dat is een persoonlijke kwestie ben ik bang. Met settings.py heb ik nog niet gewerkt. Zou je kunnen aangeven hoe je dit precies moet doen? Ik heb het Homey script inderdaad onder de juiste directory en naam gezet.

dennissiemensma, 2021-08-05:
Over het inschakelen van die plugin kun je hier meer informatie vinden:
- https://dsmr-reader.readthedocs.io/nl/v4/reference/plugins.html#configuration

Als je het script `forward_raw_telegram_to_api` hebt genoemd, dan wordt de instelling dus `DSMRREADER_PLUGINS=dsmr_plugins.modules.forward_raw_telegram_to_api`

Joriempie, 2021-08-06:
Dit staat er in de logging:

> Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-08-04 16:13:39,394 ERROR    mixins       run_once                        101 | dsmr_backend.management.commands.dsmr_backend: [!] Exception raised. Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.AdminShutdown: terminating connection due to administrator command
server closed the connection unexpectedly
        This probably means the server terminated abnormally
        before or while processing the request.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 33, in run
    dsmr_backend.services.schedule.execute_scheduled_processes()
  File "/home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py", line 28, in execute_scheduled_processes
    logger.debug('SP: %s backend service(s) ready to run', len(calls))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/query.py", line 1308, in _fetch_all
    self._result_cache = list(self._iterable_class(self))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/sql/compiler.py", line 1156, in execute_sql
    cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.OperationalError: terminating connection due to administrator command
server closed the connection unexpectedly
        This probably means the server terminated abnormally
        before or while processing the request.

Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-08-05 13:39:27,860 ERROR    mixins       run_once                        101 | dsmr_backend.management.commands.dsmr_backend: [!] Exception raised. Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.AdminShutdown: terminating connection due to administrator command
server closed the connection unexpectedly
        This probably means the server terminated abnormally
        before or while processing the request.

Joriempie, 2021-08-06:
> Over het inschakelen van die plugin kun je hier meer informatie vinden:
> 
> * https://dsmr-reader.readthedocs.io/nl/v4/reference/plugins.html#configuration
> 
> Als je het script `forward_raw_telegram_to_api.py` hebt genoemd, dan wordt de instelling dus
> 
> ```
> DSMRREADER_PLUGINS=dsmr_plugins.modules.forward_raw_telegram_to_api
> ```

Check. Ik weet al wat mijn probleem is. Ik dacht dat ik een nieuw bestand moest maken met bijvoorbeeld:
`forward_raw_telegram_to_api.env`

Maar dat blijkt niet zo te zijn. Want er staat binnen de hoofd directory: "home/dsmr/dsmr-reader" een `.env` file echter zag ik deze niet staan omdat ik `ls` heb gebruikt in plaats van `ls -la`.

Ik kwam er achter doordat ik debugging aan wou zetten via deze link: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html echter zag ik:
`### Logging level.
###DSMRREADER_LOGLEVEL=DEBUG` niet staan waarbij een lampje bij mij ging branden. Doormiddel van het commando: `find . -type f -name "*.env"` kwam ik er achter waar het bestandje vond.

closed 2021-08-06 by Joriempie

dennissiemensma, 2021-08-06:
@Joriempie fijn om te horen! Wellicht dat het laden van plugins in een toekomstige versie wat makkelijker wordt, maar voor nu is het even puzzelen soms.