dsmrreader/dsmr-reader #1430: Is MQTT : dsmr/day-consumption/energy_supplier_price_gas geen float?

bertdeelman, 2021-08-06:
Hi,

Ik gebruik Telegraf om de MQTT naar InfluxDB te sturen. Zo kan ik meer waardes gebruiken voor een Grafana dashboard dan met de feature binnen DSMR om data rechtstreeks naar InfluxDB te sturen. 

De configuratie van Telegraf werkt. Alleen krijg ik een foutmelding op dsmr/day-consumption/energy_supplier_price_gas. Dat blijkt geen float te zijn volgens Telegraf. De andere waardes binnen dsmr/day-consumption/# komen netjes in InfluxDB zonder foutmelding binnen Telegraf.

Iemand een idee?

Bert

dennissiemensma, 2021-08-06:
Bedankt voor je vraag. Ik ben niet bekend in Telegraf, maar ik zou verwachten dat MQTT-clients zelf type-conversie kunnen doen van waardes. 

Kun je wellicht ergens zoiets gebruiken? Die cast allerlei types naar floats: https://docs.influxdata.com/influxdb/v2.0/reference/flux/stdlib/built-in/transformations/type-conversions/tofloat/

bertdeelman, 2021-08-07:
Bedankt Dennis voor je snelle antwoord. Alle andere waardes uit dsmr/day-consumption/# zijn wel floats. Ook bijvoorbeeld dsmr/day-consumption/energy_supplier_price_electricity_delivered_1. Ook een waarde die je handmatig hebt ingevoerd binnen DSMR. Daarom mijn vraag. 

Ik ga kijken naar het omzetten van de gasprijs naar een float.

dennissiemensma, 2021-08-07:
Ik heb er verder ook geen verklaring voor. Helemaal omdat het interne type in DSMR-reader al een Decimal is. Laat maar weten of het zelf omzetten werkbaar is.

bertdeelman, 2021-08-10:
Opgelost. In de admin van de reader stond een vinkje bij MQTT JSON. Dat vond telegraf niet prettig.

closed 2021-08-10 by bertdeelman

dennissiemensma, 2021-08-10:
Fijn om te horen dat je het gevonden hebt!