dsmrreader/dsmr-reader #1432: Krijg dsmr niet meer werkend, HELP!!!

newtech2015, 2021-08-07:
Hallo,

Vanmorgen, probleem gehad met een sambo share, die op dezelfde externe hdd draaid als de dsmr database.

Toen dit verholpen was, nog even dsmr geupdate naar de laatste nieuwe versie, en een nieuw sd kaart image gemaakt.

hierna, werkte dsmr niet meer, en krijg het niet meer opgelost.
ook een eerder gemaakt image, krigt dsmr niet meer in de lucht.

updaten van dsmr, of herstellen, met:
sudo su - dsmr
export SETUPTOOLS_USE_DISTUTILS=stdlib
./deploy.sh

resulteerd in honderden errors waar ik niets mee kan.

gezien, t ook niet werkt, met een ouder sd kaart image, zit ik nu te denken, dat wellicht de database op de externe hdd niet goed is, of mss de hele hdd??

echter de procedure die ik volg om de backup van afgelopen nacht terug te zetten:
supervisorctl stop all
sudo systemctl stop postgresql  (stop de database)
sudo sudo -u postgres dropdb dsmrreader (verwijderen database)
sudo sudo -u postgres createdb -O dsmrreader dsmrreader (aanamken database zonder inhout)
zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Friday.sql.gz | sudo sudo -u postgres psql dsmrreader

geeft ook al heel vlug een error:
sudo sudo -u postgres dropdb dsmrreader
could not change directory to "/root": Permission denied
dropdb: could not connect to database template1: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?

klopt deze procedure niet meer, omdat deze wellicht voor database restore op interne opslag is, of wat is er aan de hand?

ik kom er niet meer uit, graag wat hulp

gr ron

dennissiemensma, 2021-08-07:
Bedankt voor je vraag. Als je fouten krijgt is het goed om te delen. Krijg je bij het deployen ook dezelfde fout?

```
Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?
```

Het lijkt er namelijk op dat je database niet werkt. Je kunt dit doen:
```
sudo service postgresql status
```

En als die niet draait:
```
sudo service postgresql restart
```

newtech2015, 2021-08-07:
hoi,
● postgresql.service - PostgreSQL RDBMS
   Loaded: loaded (/lib/systemd/system/postgresql.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sat 2021-08-07 13:53:52 CEST; 4h 29min ago
  Process: 483 ExecStart=/bin/true (code=exited, status=0/SUCCESS)
 Main PID: 483 (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/postgresql.service

Aug 07 13:53:52 raspberrypi systemd[1]: Starting PostgreSQL RDBMS...
Aug 07 13:53:52 raspberrypi systemd[1]: Started PostgreSQL RDBMS.

staat started, dus,  lijkt te werken.
tijdens een deply, gaat t een tijd goed, tot ie applying migrations gaat doen:
 --- Applying database migrations.
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/psycopg2/__init__.py", line 127, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: could not connect to server: Connection refused
        Is the server running on host "localhost" (127.0.0.1) and accepting
        TCP/IP connections on port 5432?
could not connect to server: Cannot assign requested address
        Is the server running on host "localhost" (::1) and accepting
        TCP/IP connections on port 5432?

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/management/base.py", line 85, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/management/commands/migrate.py", line 92, in handle
    executor = MigrationExecutor(connection, self.migration_progress_callback)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/migrations/executor.py", line 18, in __init__
    self.loader = MigrationLoader(self.connection)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/migrations/loader.py", line 53, in __init__
    self.build_graph()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/migrations/loader.py", line 216, in build_graph
    self.applied_migrations = recorder.applied_migrations()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/migrations/recorder.py", line 77, in applied_migrations
    if self.has_table():
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/migrations/recorder.py", line 55, in has_table
    with self.connection.cursor() as cursor:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 235, in _cursor
    self.ensure_connection()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
[... 152 lines cut ...]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
[... cut ...]
        - Create an issue on GitHub and attach the exeception trace listed above.

maar, nu ik t allemaal eens heb laten bezinken, lijkt t mij toch dat t in de database moet zitten, ik heb immers de laatste 2 complete sd-kaart images op een andere sd kaart gezet,... dan zou t gewoon weer moeten werken.enige wat dan nog "oud is",  is de database / de externe hdd.
tevens, kan ik de database niet terug zetten,.. wijst ook op probleem met database.
weet je nog iets anders om toegang te forceren, dat ik mn database backup terug kan zetten?
gr ron

dennissiemensma, 2021-08-07:
Kun je wel in de database komen met:
```
sudo sudo -u postgres psql --list
```

Het is wel vreemd dat die aangeeft dattie draait, maar DSMR-reader er dan niet bij kan.

newtech2015, 2021-08-07:
hoi,
laat maar... heb de fout gevonden:
ivm samba probleem, had ik de rechten opnieuw ingesteld van de share...
maar in een submap hiervan, zit de database van dsmr, nu de rechten van de database gezet, zoals t hoort, en t werkt gelukkig weer.
gr ron

dennissiemensma, 2021-08-07:
Oke fijn om te horen!

closed 2021-08-07 by dennissiemensma