dsmrreader/dsmr-reader #1433: MQTT messages not sent

TheMClaeys, 2021-08-09:
I tried setting up MQTT to send usage data to Home Assistant.

However, I don't receive anything on Home Assistant and the message queue is always empty.

```

DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                             "None" / "101"

DATA
    Telegrams total (est.)                                                                             408389
    Consumption records electricity / gas (est.)                                                46704 / 14166

```

The connection itself works, I changed the name to test and Mosquito logged the following:
```
1628492672: Client OurDSMR disconnected.
1628492673: New connection from 192.168.1.11 on port 1883.
1628492673: New client connected from 192.168.1.11 as DSMR (p2, c1, k60, u'<username>').
...
1628492794: Client DSMR disconnected.
...
1628492836: New connection from 192.168.1.11 on port 1883.
1628492836: New client connected from 192.168.1.11 as DSMR (p2, c1, k60, u'<username>').
```

Since nothing is received, no entities are created so no data can be used.

Supervisorctl status shows everything running fine since the last update (21 days ago, no idea if it was working before that).
Restart didn't fix it.
Backend log shows the following (multiple times):
`
2021-08-09 08:36:06,970 ERROR    persistent_clients run                                                                                                                               37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...`

After putting debug logging on I see one message with values, however nothing is seen in the logs of Mosquito: (placed in a clickable block due to length).

<details>

```
root@dsmr:/# tail /var/log/supervisor/dsmr_backend.log
2021-08-09 09:21:00,628 DEBUG    broker       run                             10                                                                                                 5 | MQTT: Deleting published message (#1496714) from queue
2021-08-09 09:21:00,628 DEBUG    broker       run                              7                                                                                                 8 | MQTT: Publishing queued message (#1496713) for dsmr/day-consumption: {"elect                                                                                                 ricity1": "0.860", "electricity2": "2.730", "electricity1_returned": "0.170", "e                                                                                                 lectricity2_returned": "0.000", "electricity_merged": "3.590", "electricity_retu                                                                                                 rned_merged": "0.170", "electricity1_cost": "0.27", "electricity2_cost": "0.87",                                                                                                  "electricity_cost_merged": "1.14", "total_cost": "1.14", "gas": "0.060", "gas_c                                                                                                 ost": "0.00", "fixed_cost": "0.00", "energy_supplier_price_electricity_delivered                                                                                                 _1": "0.317700", "energy_supplier_price_electricity_delivered_2": "0.317700", "e                                                                                                 nergy_supplier_price_electricity_returned_1": "0.045450", "energy_supplier_price                                                                                                 _electricity_returned_2": "0.045450", "energy_supplier_price_gas": "0.050090"}
2021-08-09 09:21:00,629 DEBUG    broker       on_log                          15                                                                                                 8 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m29), 'b'dsmr/day-consumpti                                                                                                 on'', ... (645 bytes)
2021-08-09 09:21:00,629 DEBUG    broker       run                              9                                                                                                 7 | MQTT: Waiting for message (#1496713) to be marked published by broker
2021-08-09 09:21:00,629 DEBUG    broker       on_log                          15                                                                                                 8 | MQTT: (Paho on_log) Received PUBREC (Mid: 29)
2021-08-09 09:21:00,629 DEBUG    broker       on_log                          15                                                                                                 8 | MQTT: (Paho on_log) Sending PUBREL (Mid: 29)
2021-08-09 09:21:00,629 DEBUG    broker       run                              9                                                                                                 7 | MQTT: Waiting for message (#1496713) to be marked published by broker
2021-08-09 09:21:00,629 DEBUG    broker       on_log                          15                                                                                                 8 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 29)
2021-08-09 09:21:00,629 DEBUG    broker       run                             10                                                                                                 5 | MQTT: Deleting published message (#1496713) from queue
2021-08-09 09:21:00,631 DEBUG    mixins       run_loop                         7                                                                                                 3 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
root@dsmr:/# tail /var/log/supervisor/dsmr_backend.log
2021-08-09 09:21:05,663 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-08-09 09:21:06,666 DEBUG    schedule     execute_scheduled_processes      28 | SP: 0 backend service(s) ready to run
2021-08-09 09:21:06,668 DEBUG    persistent_clients run                              30 | CLIENTS: Running 1 active client(s)
2021-08-09 09:21:06,670 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-08-09 09:21:07,672 DEBUG    schedule     execute_scheduled_processes      28 | SP: 0 backend service(s) ready to run
2021-08-09 09:21:07,675 DEBUG    persistent_clients run                              30 | CLIENTS: Running 1 active client(s)
2021-08-09 09:21:07,676 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-08-09 09:21:08,679 DEBUG    schedule     execute_scheduled_processes      28 | SP: 0 backend service(s) ready to run
2021-08-09 09:21:08,681 DEBUG    persistent_clients run                              30 | CLIENTS: Running 1 active client(s)
2021-08-09 09:21:08,682 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
root@dsmr:/# tail /var/log/supervisor/dsmr_backend.log
2021-08-09 09:21:28,815 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-08-09 09:21:29,818 DEBUG    schedule     execute_scheduled_processes      28 | SP: 0 backend service(s) ready to run
2021-08-09 09:21:29,820 DEBUG    persistent_clients run                              30 | CLIENTS: Running 1 active client(s)
2021-08-09 09:21:29,821 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-08-09 09:21:30,824 DEBUG    schedule     execute_scheduled_processes      28 | SP: 1 backend service(s) ready to run
[... 207 lines cut ...]
2021-08-09 09:38:28,456 DEBUG    mixins       run_loop                         73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-08-09 09:38:29,458 DEBUG    schedule     execute_scheduled_processes      28 | SP: 0 backend service(s) ready to run
2021-08-09 09:38:29,460 DEBUG    persistent_clients run                              30 | CLIENTS: Running 1 active client(s)
[... cut ...]
The error doesn't show up when running in debug (or it takes a log longer, however nothing was received on Mosquito yet).

What more info can/should I provide to help troubleshoot this?
(I am also not 100% sure it's an issue on DSMR, but since it shows the error in the log and the connection to MQTT itself seems to work I ask the question here).

Both systems are in the same subnet, running on the same ESX host.
It worked before on a previous install of DSMR, but I migrated a while back from a Pi to an ESX system and I didn't really test it since.

dennissiemensma, 2021-08-09:
Thanks for asking and providing quite some information.

It *seems* that the problem is between the broker and HA. The logs state the following:

- You are using [MQTT QoS 2 level](https://www.hivemq.com/blog/mqtt-essentials-part-6-mqtt-quality-of-service-levels/) (the most reliable), which requires both the MQTT client (DSMR-reader) and the broker to confirm communication twice. It also ensures each MQTT message is sent/processed exactly _once_.
- Samples:
```
(Paho on_log) Sending PUBLISH (d0, q2, r1, m29), 'b'dsmr/day-consumpti                                                                                                 on'', ... (645 bytes)
(Paho on_log) Received PUBREC (Mid: 29)
(Paho on_log) Sending PUBREL (Mid: 29)
(Paho on_log) Received PUBCOMP (Mid: 29)
```

The `PUBREC` and `PUBCOMP` are confirmations of the broker receiving the message, so I would expect the broker to receive them properly.

Note that DSMR-reader *should* sent MQTT messages with the `retain` flag as well, meaning that the broker *should* keep track of the latest message of each topic and re-send it to any new clients (such as HA), even if they were not listening the moment the (latest) MQTT message was received by the broker.

---

- You can scan/grep the `Sending PUBLISH...` messages to see whether the topics you expect are actually sent (and confirmed). E.g.: `'b'dsmr/day-consumption''` of the sample above.
- You may be able to provide some details of logs around `Client loop()` (I cannot find them in the log above) so check their root cause.

TheMClaeys, 2021-08-10:
Thanks for the reply. I think QoS2 is the only allowed one since the last update.

That "Client loop() failed" log only shows up apparently when debug logging isn't on.
Probably because it's only a few times a day.

```
2021-08-08 07:56:09,887 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 08:31:10,350 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 08:36:13,036 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 08:39:14,884 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 09:34:09,640 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 14:32:06,825 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 14:36:09,122 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 14:41:11,849 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 14:45:14,732 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 18:29:08,759 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 18:32:10,676 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 18:59:08,093 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 19:14:15,136 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 19:27:06,181 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 19:40:12,347 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 20:01:10,274 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-08 20:10:14,762 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-09 08:02:05,890 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-09 08:16:12,487 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-09 08:30:03,821 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-09 08:36:06,970 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-09 08:41:09,663 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-09 12:04:02,061 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-09 14:31:07,477 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-09 17:21:06,529 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-09 18:29:07,382 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-10 08:09:13,487 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-10 08:29:07,572 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-10 08:34:10,316 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-10 08:38:12,613 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2021-08-10 08:45:01,091 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
```
It doesn't provide more info than that, however I found 1 line when I grep the last 100k lines that is different:
`2021-07-28 20:30:05,142 ERROR    persistent_clients run                              37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart to prevent waiting forever...`

[... cut ...]
2021-08-10 09:14:00,299 DEBUG    broker       on_log                          158 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m121), 'b'dsmr/day-consumption'', ... (645 bytes)
2021-08-10 09:15:00,707 DEBUG    messages     queue_message                    36 | MQTT: Rejecting message as it exactly matches the previous message sent for this topic: dsmr/day-consumption
```

I don't think I can set extra logging on Mosquito so I'm a bit blind on that side, I think it shows every log it receives, at least it used to do that.

dennissiemensma, 2021-08-10:
Those errors are (likely) related to network issues:
- https://github.com/dsmrreader/dsmr-reader/blob/v4.16.3/dsmr_mqtt/services/broker.py#L91
- https://github.com/dsmrreader/dsmr-reader/blob/v4.16.3/dsmr_mqtt/services/broker.py#L103

But I'm not sure, since it maps any non-success result to the error.

Even so, if the errors do not persist every second (since there seems to be quite some time between most of them) it's probably not the root cause, just a side error.

----

Regarding the data in the broker, have you tried subscribing to all topics to see whatever is available?

E.g. on the host of the broker:
```
mosquitto_sub -d -t '#'
```
For me (with a single topic tested) it returns:

```
Client (null) sending CONNECT
Client (null) received CONNACK (0)
Client (null) sending SUBSCRIBE (Mid: 1, Topic: #, QoS: 0, Options: 0x00)
Client (null) received SUBACK
Subscribed (mid: 1): 0
Client (null) received PUBLISH (d0, q0, r1, m0, 'dsmr/json', ... (798 bytes))
{"id": 5, "timestamp": "2021-08-10T19:34:00+02:00", "electricity_delivered_1": "20854.644", "electricity_returned_1": "0.000", "electricity_delivered_2": "12512.787", "electricity_returned_2": "0.000", "electricity_currently_delivered": "0.000", "electricity_currently_returned": "3.381", "phase_currently_delivered_l1": "0.000", "phase_currently_delivered_l2": "0.000", "phase_currently_delivered_l3": "0.000", "extra_device_timestamp": "2021-08-10T19:30:00+02:00", "extra_device_delivered": "6256.394", "phase_currently_returned_l1": "1.069", "phase_currently_returned_l2": "1.050", "phase_currently_returned_l3": "1.262", "phase_voltage_l1": "255.6", "phase_voltage_l2": "257.6", "phase_voltage_l3": "262.6", "phase_power_current_l1": 5, "phase_power_current_l2": 3, "phase_power_current_l3": 1}
```

The message was sent a few seconds before subscribing, but Mosquitto seems to cache it properly due to the `retain` flag. 

Do you see similar output with the command above?

TheMClaeys, 2021-08-10:
I'll have to check if I can do that, it's an integration in Home Assistant within its own docker container I think. I'll check it tomorrow to see if I can figure that out.

dennissiemensma, 2021-08-10:
You can try any other client though. It's just that the example above seemed the easiest one to do, were you to have easy access to the host where the server is installed. :see_no_evil: 

It should also work on any other command line where the mosquitto software (and therefore `mosquitto_sub`) is installed. Maybe a random MQTT explorer works as well.

TheMClaeys, 2021-08-11:
I learned some new things today :flushed:

Anyway:
```
Message 279 received on dsmr/day-consumption at 08:36:

{
    "electricity1": "0.867",
    "electricity2": "2.761",
    "electricity1_returned": "0.000",
    "electricity2_returned": "0.000",
    "electricity_merged": "3.628",
    "electricity_returned_merged": "0.000",
    "electricity1_cost": "0.28",
    "electricity2_cost": "0.88",
    "electricity_cost_merged": "1.16",
    "total_cost": "1.17",
    "gas": "0.105",
    "gas_cost": "0.01",
    "fixed_cost": "0.00",
    "energy_supplier_price_electricity_delivered_1": "0.317700",
    "energy_supplier_price_electricity_delivered_2": "0.317700",
    "energy_supplier_price_electricity_returned_1": "0.045450",
    "energy_supplier_price_electricity_returned_2": "0.045450",
    "energy_supplier_price_gas": "0.050090"
}

QoS: 0 - Retain: false

Message 278 received on dsmr/current-period at 08:36:

{
    "current_month_electricity1": "20.506",
    "current_month_electricity1_cost": "1.23",
    "current_month_electricity1_returned": "116.374",
    "current_month_electricity2": "57.653",
    "current_month_electricity2_cost": "16.66",
    "current_month_electricity2_returned": "36.474",
    "current_month_electricity_merged": "78.159",
    "current_month_electricity_cost_merged": "17.89",
    "current_month_electricity_returned_merged": "152.848",
    "current_month_gas": "3.889",
    "current_month_gas_cost": "0.18",
    "current_month_fixed_cost": "0.00",
    "current_month_total_cost": "18.07",
    "current_year_electricity1": "329.639",
    "current_year_electricity1_cost": "19.45",
    "current_year_electricity1_returned": "1876.420",
    "current_year_electricity2": "861.892",
    "current_year_electricity2_cost": "241.71",
    "current_year_electricity2_returned": "705.547",
    "current_year_electricity_merged": "1191.531",
    "current_year_electricity_cost_merged": "261.16",
    "current_year_electricity_returned_merged": "2581.967",
    "current_year_gas": "250.424",
    "current_year_gas_cost": "12.57",
    "current_year_fixed_cost": "0.00",
    "current_year_total_cost": "273.73"
}

QoS: 0 - Retain: false

```

So messages are coming in apparently.
QoS isn't version 2 in the messages but I don't know if that matters.

Since the message comes in I guess that means it's a Home Assistant issue?

dennissiemensma, 2021-08-11:
I'm closing the issue, but feel free to ask any follow up questions. Even if I might not have the answer you're looking for.

closed 2021-08-11 by dennissiemensma