dsmrreader/dsmr-reader #1434: Feature request: Netto teruglevering via MQTT

bobvmierlo, 2021-08-10:
Het zou fijn zijn als DSMR Reader de netto teruglevering berekend en deze via MQTT kan uitleveren aan bijvoorbeeld Home Assistant. Nu krijgen we alleen de teruglevering en het verbruik, maar niet het netto verschil.
Het nieuwe Energy dashboard van Home Assistant kan dan een percentage tonen van zelf gebruikte zonne-energie.

Mijn DSMR debug dump:
```
DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.9.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                    1.0s / 10.0s / 168h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                              34810
    Consumption records electricity / gas (est.)                                                 22245 / 5873
```

dennissiemensma, 2021-08-10:
Bedankt voor je verzoek. Het hangt er denk ik vanaf wat je exact nodig hebt, wat DSMR-reader momenteel nog niet ontsluit. 

Als het puur om het _verschil_ tussen het verbruik en teruglevering gaat, zoals de slimme meter het 1-op-1 doorgeeft, is dat denk ik iets wat het beste aan dat kant van HA (of de software die het uit de MQTT-broker haalt) gedaan kan worden. Bijvoorbeeld in de bijbehorende integratie:
- https://github.com/home-assistant/core/labels/integration%3A%20dsmr_reader

Het is namelijk aan de kant van DSMR-reader onwerkbaar om voor elke combinatie of operatie van velden een aparte waarde te maken, die te ontsluiten en ook voor altijd te blijven onderhouden.

Laat maar weten!

bobvmierlo, 2021-08-11:
Het enige nadeel van zelf aanmaken in HA is dat dit met 2 tussenstappen moet, de standaard 'template sensor' bevat namelijk niet de sensor attributen die nodig zijn om de sensor te kunnen gebruiken in het nieuwe 'Energy dashboard'. Daarna moet je weer een 'Utility meter' sensor aanmaken met de data uit de template sensor. Aangezien de data die vanuit de DSMR Reader MQTT berichten binnenkomt door de HA integration automatisch de juiste sensor attributen krijgt was dit voor mij de simpelste oplossing.

Het is hoe dan ook gelukt met bovenstaande workaround, toch bedankt voor het meedenken!

closed 2021-08-11 by bobvmierlo

dennissiemensma, 2021-08-11:
Helder, dank voor de terugkoppeling!

Ik snap dat het wellicht wat lastig is, maar anderzijds zie ik ook wel kansen voor HA om hier iets voor te maken. Helemaal als dat nieuwe dashboard een goede uitkomst is voor het gros van de gebruikers daar. Want DSMR-reader is uiteraard niet de enige databron die gemapped moet worden, dus die vraag komt daar vast nog wel langs voor een hoop andere bronnen. 

En de grafische kant van DSMR-reader is eigenlijk ook niets meer dan "een" intepretatie van de metergegevens, net zoals dat HA-dashboard, dus vandaar dat ik het liever generiek houdt via de API/MQTT, zodat iedereen er zelf op kan voortbouwen waar nodig.