dsmrreader/dsmr-reader #1435: mqtt foutmeldingen

Jos-77, 2021-08-12:
ls,

ik wil dsrm reader koppelen aan Home Assistant, ik heb een goed draaiende configuratie met Mosquito mqtt.

ik krijg de volgende foutmelding.

1628749750: New connection from 172.30.32.2 on port 1883.
1628749750: Socket error on client <unknown>, disconnecting.

instellingen dsmr mqtt

hostnaam: ipadres home assistant
poort 1883
beveiliging onveilig 
clientid: DVES_4RPI3B

gebruikersnaam: gebruikersnaam mosquito 
wachtwoord: wachtwoord mosquito

iemand enig idee?

dennissiemensma, 2021-08-12:
Bedankt voor je vraag. Het is mij niet helemaal duidelijk wat er wel werkt en wat niet.

- Heb je ook al in de logs van `dsmr_backend` gekeken? https://dsmr-reader.readthedocs.io/nl/v4/how-to/troubleshooting/logfiles.html

dennissiemensma, 2021-08-17:
@Jos-77 ben je hier nog verder mee gekomen?

Jos-77, 2021-08-17:
Nee, heb mijn server in de meterkast gezet, geen mqtt meer nodig.

Het zit ergens in de settings, maar niet te vinden, wel fout logs van ha, niets van dsmr.

dennissiemensma, 2021-08-17:
Oke bedankt voor de update! Mocht je later nog ergens anders tegen aanlopen, laat tzt maar weten!

closed 2021-08-17 by dennissiemensma