dsmrreader/dsmr-reader #1436: dsmr webinterface reports blank...

JeeKee80, 2021-08-15:
Hi guys,

I am kinda lost, but my dsmr webinterface gives me blank when accessing it with iOS safari\chrome or directly from my pi. If I access the webinterface with my mac it show as it should. I am using NGINX as server running on a pi to only offer local access. dsmr is up-to-date, python 3.7 and pip 21.2.

DSMR webinterface is running as it should. If I use curl I always see a GET request. No errors\warnings in the NGINX logs.

Any ideas?

dennissiemensma, 2021-08-15:
You may want to start with pressing F12 or whatever brings up the developer tools in de browser. Then check the Console tab for any errors and the Network tab to see what status code the requests return. Either may give more information about why it fails.

[image]

JeeKee80, 2021-08-15:
Thanks, but also there it shows only the blank part without any gets. Seems like the gets are being dropped when sent by any ios browser but not through mac browser. Funny thing is that the curl requests do come through from both devices. So no firewall issues or anything. Although i did notice some code within dsmr script about standalone (only applicable to safari)? But still weird that I would be the only one experiencing this…

dennissiemensma, 2021-08-15:
Unfortunately I do not have an iOS device to test with, so you'll need to be really specific. From a technical point of view, I see no obvious reason for requests to be handled differently. 

> Thanks, but also there it shows only the blank part without any gets. 

If requests were sent but dropped by the server, you should see failing requests. If no requests are sent at all, in two browsers on the same device, it's gonna be really hard to debug those.

To be clear, you did try the developer tools on the iOS device? https://developer.apple.com/safari/tools/

----

> Although i did notice some code within dsmr script about standalone (only applicable to safari)

Could you point to the code so I can check out if it's related or not?

----

> I am using NGINX as server running on a pi to only offer local access. 

Is this the webserver serving DSMR-reader or is it another proxy in front of it?

----

If you can't find any other clues, you may want to debug the Nginx server to see whether it receives any request at all from the browsers, and therefor whether the browsers are causing the issue or something else. 
I'm not entirely sure what the best way of debugging would be. You could either try debug logging or use `tcpdump` on the server running Nginx and see over there whether any requests from the browsers come in.

JeeKee80, 2021-08-16:
> Assigned #1436 <https://github.com/dsmrreader/dsmr-reader/issues/1436> to @JeeKee80 <https://github.com/JeeKee80>.

Ok, I have found the culprit. Apparently port 10080 (the one that I was using as an alternative to port 80) has been silently closed on iOS Safari or so it seems (following Chrome and Firefox I suppose). Although I can’t find any reference from apple on the matter (although I am running an iOS beta, but strangely enough my other iPhone runs non-beta iOS and that one as well blocks port 10080. So changing the port solved the problem for me!

Thanks for helping out @Dennis Siemensma

dennissiemensma, 2021-08-16:
Great and thanks for sharing the info as well!

closed 2021-08-16 by dennissiemensma