dsmrreader/dsmr-reader #1437: Server Error DSMR-reader

Subz3r00, 2021-08-16:
Hello there,

I got the following error:

Server Error
Sorry, something unexpected happened.
Exception:
OperationalError: could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/handlers/base.py", line 181, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/mixins.py", line 10, in dispatch
    if not FrontendSettings.get_solo().always_require_login:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/solo/models.py", line 55, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/query.py", line 573, in get_or_create
    return self.get(**kwargs), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/query.py", line 425, in get
    num = len(clone)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/query.py", line 1308, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/models/sql/compiler.py", line 1154, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 235, in _cursor
    self.ensure_connection()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)

  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)

I check the following things:

- DB (PostgreSQL) is running fine 
- supervisorctl status is just fine running how it should be.
- Full disk not the issue only 32% full 

What stuff do i use:

- it's an Ubunto machine in docker 
- Version 4.16.3 

If need more information i will provide.

Subz3r00, 2021-08-16:
It has something to do with the DB is guess, because if i use the ./deploy.sh i get the error below

 >>>>> [!] Executing database migrations failed! <<<<<
       [i] Trying to automatically resolve with 'dsmr_sqlsequencereset'.
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: could not connect to server: Connection refused
        Is the server running on host "localhost" (127.0.0.1) and accepting
        TCP/IP connections on port 5432?

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_sqlsequencereset.py", line 29, in handle
    with connection.cursor() as cursor:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 235, in _cursor
    self.ensure_connection()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
django.db.utils.OperationalError: could not connect to server: Connection refused
        Is the server running on host "localhost" (127.0.0.1) and accepting
        TCP/IP connections on port 5432?

Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 200, in connect
[... 44 lines cut ...]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/utils.py", line 90, in __exit__
[... cut ...]
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
django.db.utils.OperationalError: could not connect to server: Connection refused
        Is the server running on host "localhost" (127.0.0.1) and accepting
        TCP/IP connections on port 5432?

>>>>> [!] Executing database migrations failed again! <<<<<
        - Running dsmr_sqlsequencereset did not resolve the problem.
        - Create an issue on GitHub and attach the exeception trace listed above

dennissiemensma, 2021-08-16:
Hi, DSMR-reader is unable to connect to the database. If you are not running it on the same host/container, you will need to change the hostname used by the database client (DSMR-reader). See:
- https://dsmr-reader.readthedocs.io/en/v4/reference/env-settings.html#django-database-host

If the DB can then be reached, but DSMR-reader is still unable to connect, the error message should at least change.

You can also try `telnet` (if installed on the server/container running DSMR-reader) to see if the connection can be established at all, e.g.:
```
telnet 127.0.0.1 5432
```

Subz3r00, 2021-08-16:
Hi Dennis,

When i try to telnet to 127.0.0.1 and port 5432 the connection is refused.

dennissiemensma, 2021-08-16:
Are you running the database in Docker or did you install it directly into Ubuntu? Are you using docker-compose? Where is the DB running and where DSMR-reader?

Subz3r00, 2021-08-16:
I made a mistake to say my DSMR was in docker my HA is in docker, but DSMR is just a vitrual machine with ubuntu and i installed the docker on the Ubuntu machine.

I followed this guide https://dsmr-reader.readthedocs.io/nl/v4/how-to/installation/quick.html 

So the DB running where die DSMR-reader is.

dennissiemensma, 2021-08-16:
Are you sure PostgreSQL is running? If they are installed on the same host, the connection should be fine. If either runs on another host, then you'll need to allow "external" connections to the DB.

Try:
```
sudo service postgresql status
```

Sample:
```
sudo service postgresql status
● postgresql.service - PostgreSQL RDBMS
   Loaded: loaded (/lib/systemd/system/postgresql.service; enabled; vendor preset: enabled)
   Active: active (exited) since Wed 2021-06-09 22:16:38 CEST; 2 months 7 days ago
  Process: 16252 ExecStart=/bin/true (code=exited, status=0/SUCCESS)
 Main PID: 16252 (code=exited, status=0/SUCCESS)

Jun 09 22:16:38 rpi4.x.net systemd[1]: Starting PostgreSQL RDBMS...
Jun 09 22:16:38 rpi4.x.net systemd[1]: Started PostgreSQL RDBMS.
```

Check whether the DB port is bound and accepting connections:
```
sudo netstat -plant | grep 5432
```

E.g.:
```
sudo netstat -plant | grep 5432
tcp        0      0 127.0.0.1:5432          0.0.0.0:*               LISTEN      16230/postgres      
tcp6       0      0 ::1:5432                :::*                    LISTEN      16230/postgres      
tcp6       0      0 ::1:5432                ::1:43630               ESTABLISHED 21436/postgres: 11/ 
tcp6       0      0 ::1:43634               ::1:5432                ESTABLISHED 5113/python3        
tcp6       0      0 ::1:43632               ::1:5432                ESTABLISHED 5111/python3        
tcp6       0      0 ::1:5432                ::1:52050               ESTABLISHED 26464/postgres: 11/ 
tcp6       0      0 ::1:52050               ::1:5432                ESTABLISHED 5121/python3        
tcp6       0      0 ::1:5432                ::1:43632               ESTABLISHED 21437/postgres: 11/ 
tcp6       0      0 ::1:52108               ::1:5432                ESTABLISHED 5117/python3        
tcp6       0      0 ::1:5432                ::1:43634               ESTABLISHED 21438/postgres: 11/ 
tcp6       0      0 ::1:5432                ::1:52108               ESTABLISHED 26498/postgres: 11/ 
tcp6       0      0 ::1:5432                ::1:43636               ESTABLISHED 21439/postgres: 11/ 
tcp6       0      0 ::1:43636               ::1:5432                ESTABLISHED 5110/python3        
tcp6       0      0 ::1:43630               ::1:5432                ESTABLISHED 5114/python3 
```

Subz3r00, 2021-08-16:
Yes, for sure is running 

When i sudo service postgresql status the following shows up.

● postgresql.service - PostgreSQL RDBMS
     Loaded: loaded (/lib/systemd/system/postgresql.service; enabled; vendor pr>
     Active: active (exited) since Mon 2021-08-16 16:04:47 CEST; 4h 57min ago
   Main PID: 616 (code=exited, status=0/SUCCESS)
      Tasks: 0 (limit: 2230)
     Memory: 0B
     CGroup: /system.slice/postgresql.service

aug 16 16:04:46 dsmr-Virtual-Machine systemd[1]: Starting PostgreSQL RDBMS...
aug 16 16:04:47 dsmr-Virtual-Machine systemd[1]: Finished PostgreSQL RDBMS.
lines 1-10/10 (END)

When i try sudo netstat -plant | grep 5432 it gives no result, but when i do sudo netstat -plat the following shows up and i don't see any port 5432 between it.

Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      -
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      -
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      -
tcp        0    816 192.168.2.13:22         192.168.2.172:54775     ESTABLISHED -
tcp        0      0 192.168.2.13:59890      35.224.170.84:80        TIME_WAIT   -
tcp6       0      0 :::22                   :::*                    LISTEN      -
tcp6       0      0 ::1:631                 :::*                    LISTEN

dennissiemensma, 2021-08-16:
Are you sure that you used `sudo` in the last command? Omitting it *may* not show all processes.
```
sudo netstat -plant
```

Anyway, if it doesn't show up it's expected, since no connection could be made anyway. Next step is checking whether the DB actually runs, despite the service status.
```
sudo ps faux | grep postgres
```

E.g.:

```
sudo ps faux | grep postgres
pi       26958  0.0  0.0   7352   484 pts/0    S+   21:30   0:00              \_ grep --color=auto postgres
postgres 16230  0.0  1.1 204048 22216 ?        S    Jun09  11:04 /usr/lib/postgresql/11/bin/postgres -D /var/lib/postgresql/11/main -c config_file=/etc/postgresql/11/main/postgresql.conf
postgres 16232  0.0  5.5 204172 106916 ?       Ss   Jun09   5:44  \_ postgres: 11/main: checkpointer   
postgres 16233  0.0  0.4 204048  9388 ?        Ss   Jun09   2:53  \_ postgres: 11/main: background writer   
postgres 16234  0.0  0.4 204048  8132 ?        Ss   Jun09  30:20  \_ postgres: 11/main: walwriter   
postgres 16235  0.0  0.3 204448  6176 ?        Ss   Jun09   3:42  \_ postgres: 11/main: autovacuum launcher   
postgres 16236  0.0  0.1  61304  3484 ?        Ss   Jun09  16:16  \_ postgres: 11/main: stats collector   
postgres 16237  0.0  0.2 204344  4980 ?        Ss   Jun09   0:06  \_ postgres: 11/main: logical replication launcher   
postgres 21436  0.0  1.1 205696 21520 ?        Ss   Aug15   0:00  \_ postgres: 11/main: dsmrreader dsmrreader ::1(43630) idle
postgres 21437  0.0  1.0 205124 19764 ?        Ss   Aug15   0:00  \_ postgres: 11/main: dsmrreader dsmrreader ::1(43632) idle
postgres 21438  0.0  1.0 205124 19768 ?        Ss   Aug15   0:00  \_ postgres: 11/main: dsmrreader dsmrreader ::1(43634) idle
postgres 21439  0.0  0.8 205640 16472 ?        Ss   Aug15   0:00  \_ postgres: 11/main: dsmrreader dsmrreader ::1(43636) idle
postgres 26759  0.2  3.5 206428 67780 ?        Ss   21:01   0:04  \_ postgres: 11/main: dsmrreader dsmrreader ::1(52444) idle
postgres 26792  0.0  0.8 205124 16840 ?        Ss   21:10   0:01  \_ postgres: 11/main: dsmrreader dsmrreader ::1(52496) idle
```

Next, check the port set for the DB its config:
```
sudo grep port /etc/postgresql/11/main/postgresql.conf
```

E.g:
```
port = 5432				# (change requires restart)
```

Subz3r00, 2021-08-16:
It is getting me more and more confusing, because when i try the command sudo grep port /etc/postgresql/11/main/postgresql.conf  he comes with the message "No such file or directory"

dennissiemensma, 2021-08-16:
You seem to run the commands as `dsmr` user. Did you add the user to the sudoer group?

```
(dsmrreader) dsmr@dsmr-Virtual-Machine:~
```

If not, any `sudo` commands should be executed as `root` or any `sudo` user (e.g. `pi` on Raspberry Pi), just to be sure. 

Can you `logout` to switch to the sudo user or `logout` and then run `sudo -s` to become `root` instead? And then execute all commands discussed before to ensure sudo permissions were not limiting your output.

Because I see no reason why the DB should not run or cannot be accessed. In the end, try `sudo systemctl restart postgresql.service` as root or sudo user. Then try `sudo ps faux | grep postgres` again.

Subz3r00, 2021-08-16:
It makes no differents 

root@dsmr-Virtual-Machine:/home/dsmr# sudo netstat -plant | grep 5432
root@dsmr-Virtual-Machine:/home/dsmr# sudo ps faux | grep postgres
root       62360  0.0  0.0   9048   732 pts/1    S+   22:11   0:00                      \_ grep --color=auto postgres
root@dsmr-Virtual-Machine:/home/dsmr# sudo grep port /etc/postgresql/12/main/postgresql.conf
port = 5432                             # (change requires restart)
#ssl_passphrase_command_supports_reload = off
                                        # supported by the operating system:
                                        # supported by the operating system:
                                        # supported by the operating system:
                                        #   %r = remote host and port

Subz3r00, 2021-08-16:
root@dsmr-Virtual-Machine:/home/dsmr# sudo systemctl restart postgresql.service
root@dsmr-Virtual-Machine:/home/dsmr# sudo ps faux | grep postgres
root       62812  0.0  0.0   9048   724 pts/1    S+   22:14   0:00                      \_ grep --color=auto postgres
postgres   62762  0.2  1.4 218300 28592 ?        Ss   22:13   0:00 /usr/lib/postgresql/12/bin/postgres -D /var/lib/postgresql/12/main -c config_file=/etc/postgresql/12/main/postgresql.conf
postgres   62779  0.0  0.2 218300  4412 ?        Ss   22:13   0:00  \_ postgres: 12/main: checkpointer
postgres   62780  0.0  0.3 218300  6716 ?        Ss   22:13   0:00  \_ postgres: 12/main: background writer
postgres   62781  0.0  0.5 218300 10124 ?        Ss   22:13   0:00  \_ postgres: 12/main: walwriter
postgres   62782  0.0  0.3 218844  7100 ?        Ss   22:13   0:00  \_ postgres: 12/main: autovacuum launcher
postgres   62783  0.0  0.2  72932  5764 ?        Ss   22:13   0:00  \_ postgres: 12/main: stats collector
postgres   62784  0.0  0.3 218844  6768 ?        Ss   22:13   0:00  \_ postgres: 12/main: logical replication launcher
postgres   62788  0.2  1.0 219992 19740 ?        Ss   22:13   0:00  \_ postgres: 12/main: dsmrreader dsmrreader 127.0.0.1(49204) idle
postgres   62793  2.8  3.0 221556 60364 ?        Ss   22:13   0:00  \_ postgres: 12/main: dsmrreader dsmrreader 127.0.0.1(49208) idle
root@dsmr-Virtual-Machine:/home/dsmr#

dennissiemensma, 2021-08-16:
Great! Now it should run fine. If you do not see data after a few minutes, check the logs:
- https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/logfiles.html

Or perform a manual reading:
- https://dsmr-reader.readthedocs.io/en/latest/tutorial/installation/step-by-step.html#optional-your-first-reading

```
sudo apt-get install cu
sudo su - dsmr
cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q
```

This assumes you've completed the (quick) installation and added the `dsmr` user to the `dialout` group, which was one of the installation commands:
```
sudo usermod -a -G dialout dsmr
```

dennissiemensma, 2021-08-16:
Oh, don't forget to run the dpeloy script once first:
```
sudo su - dsmr
./deploy.sh
```

Assuming you finished the installation before. The migrations should now run fine there.

Subz3r00, 2021-08-17:
Everthing is working fine, but only the live "usages" and "return" are not coming up.

[image]

The ./deploy is working aswell.
When i checking the logs with it get the following results.

sudo tail -f /var/log/supervisor/dsmr_webinterface*
sudo tail -f /var/log/supervisor/dsmr_datalogger*
sudo tail -f /var/log/supervisor/dsmr_backend*

**sudo tail -f /var/log/supervisor/dsmr_webinterface***
[image]

**sudo tail -f /var/log/supervisor/dsmr_datalogger***
(dsmrreader) dsmr@dsmr-Virtual-Machine:~/dsmr-reader$ sudo tail -f /var/log/supervisor/dsmr_datalogger*
==> /var/log/supervisor/dsmr_datalogger.log <==
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html

==> /var/log/supervisor/dsmr_datalogger.log.1 <==
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
==> /var/log/supervisor/dsmr_datalogger.log.2 <==
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
==> /var/log/supervisor/dsmr_datalogger.log.3 <==
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
[... cut ...]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
django.db.utils.OperationalError: could not connect to server: Connection refused
        Is the server running on host "localhost" (127.0.0.1) and accepting
        TCP/IP connections on port 5432?

Subz3r00, 2021-08-17:
I just made a new Ubuntu server same version as before and use the guide what i showed before and now the live data is showing up.

[image]

dennissiemensma, 2021-08-17:
Awesome! I assume everything works properly now. If not, always feel free to follow up later.

closed 2021-08-17 by dennissiemensma