dsmrreader/dsmr-reader #1440: Verkeerde waardes bij export uurtotalen naar CSV

tojo2412, 2021-08-19:
Bij de export van de uurtotalen naar CSV viel het mij op dat de totalen niet (altijd) juist zijn.
Volgens de CSV-export zou ik van 6u tot 10u ongeveer 7kWh terug hebben gegeven aan het net. Als ik dezelfde tijdsrange opvraag op de omvormer van mijn zonnepanelen zouden ze "maar" 2,5kWh hebben opgebracht. (zie kolom "zon totaal")

Op de grafiek in dezelfde tijdspanne kan ik ook niet afleiden dat er 7kWh zou zijn opgebracht geweest.
Ik heb ook een export op langere tijd gedaan en dit komt meerdere keren voor : ongeveer 5% van de lijnen...

Alvast bedankt !

Groetjes,
Tom

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/tOj/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/tOj/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Hour   Start | Electricity 1 (Low Tariff) | Electricity 2 (High Tariff) | Electricity 1 Returned (Low   Tariff) | Electricity 2 Returned (High   Tariff) | Zon totaal
-- | -- | -- | -- | -- | --
2021-08-17T06:00:00+00:00 | 0 | 0 | 0,6 | 0 | 0,021
2021-08-17T07:00:00+00:00 | 0 | 0 | 1,132 | 0 | 0,301
2021-08-17T08:00:00+00:00 | 0 | 0 | 3,199 | 0 | 0,865
2021-08-17T09:00:00+00:00 | 0 | 0 | 2,704 | 0 | 1,348

</body>

</html>

[image]

dennissiemensma, 2021-08-19:
Bedankt voor je melding. Ik denk dat het twee oorzaken heeft:
- #1385 - waarbij de uurstatistieken (die in deze CSV export gebruikt worden) een uur scheef lopen
- Lokale tijdzone <> UTC - De export is nu (onbedoeld) in UTC (zie `+00:00` in je data). Hierdoor verschilt het 1 uur met wintertijd en 2 uren met zomertijd. Dit is hoe het intern opgeslagen wordt, maar niet bedoeld om ook zo te tonen aan gebruikers.

Het eerste is iets wat in de volgende release gefixt wordt (initieel wel alleen voor nieuwe data). Het tweede is ook relatief makkelijk te fixen en zal ik daarin meenemen.

----

Ter illustratie over UTC, hier een dump van mijn data rond de overgang wintertijd/zomertijd. De gegevens zijn gelijk, maar de tijdstippen niet:

Huidige versie
```
Hour Start,Electricity 1 (Low Tariff),Electricity 2 (High Tariff),Electricity 1 Returned (Low Tariff),Electricity 2 Returned (High Tariff),Gas
2021-03-27T23:00:00+00:00 ,0.266,0.0,0.0,0.0,0.0
2021-03-28T00:00:00+00:00 ,0.131,0.0,0.0,0.0,0.0
2021-03-28T01:00:00+00:00 ,0.162,0.0,0.0,0.0,0.0
2021-03-28T02:00:00+00:00 ,0.102,0.0,0.0,0.0,0.0
2021-03-28T03:00:00+00:00 ,0.142,0.0,0.0,0.0,0.0
2021-03-28T04:00:00+00:00 ,0.134,0.0,0.0,0.0,1.029
2021-03-28T05:00:00+00:00 ,0.191,0.0,0.0,0.0,1.255
2021-03-28T06:00:00+00:00 ,0.156,0.0,0.0,0.0,0.103
```

Fix:

```
Hour Start,Electricity 1 (Low Tariff),Electricity 2 (High Tariff),Electricity 1 Returned (Low Tariff),Electricity 2 Returned (High Tariff),Gas
2021-03-28T00:00:00+01:00 ,0.266,0.0,0.0,0.0,0.0
2021-03-28T01:00:00+01:00 ,0.131,0.0,0.0,0.0,0.0
2021-03-28T03:00:00+02:00 ,0.162,0.0,0.0,0.0,0.0
2021-03-28T04:00:00+02:00 ,0.102,0.0,0.0,0.0,0.0
2021-03-28T05:00:00+02:00 ,0.142,0.0,0.0,0.0,0.0
2021-03-28T06:00:00+02:00 ,0.134,0.0,0.0,0.0,1.029
2021-03-28T07:00:00+02:00 ,0.191,0.0,0.0,0.0,1.255
2021-03-28T08:00:00+02:00 ,0.156,0.0,0.0,0.0,0.103
```

dennissiemensma, 2021-08-19:
In de onderste is ook duidelijk de overgang van een uur vooruit te zien. In het voorjaar bestaat immers die nacht het tijdstip `02:00` niet:
```
2021-03-28T01:00:00+01:00
2021-03-28T03:00:00+02:00
```

tojo2412, 2021-08-19:
Hey Dennis, wat je zegt klopt volledig en kan ik ook terug vinden in mijn export. De data zou dan idd moeten opschuiven.

Volgens de omvormer zou de zonnepanelen tussen 7u en 8u 0,301 kWh hebben opgebracht, maar dsmr-reader geeft veel hogere waardes aan, zelfs na het toepassen van de winteruur-fix... 

De "returned electricity" kan toch niet hoger zijn dan wat de omvormer van de zonnepanelen heeft geproduceerd, of wel ?

dennissiemensma, 2021-08-19:
Bedankt voor je aanvulling. Ik denk dat je het beste kunt beginnen met wat de slimme meter zelf registreert. Dus mocht je morgen aan het begin en eind van een uur beschikbaar zijn, noteer even de meterstanden van teruglevering en bereken het verschil. 

Dat vergelijk je dan met je omvormer om te zien of je daar ook meer teruglevering ziet dan opwekking. Zie je dat niet, dan is de volgende stap kijken of DSMR-reader wel dezelfde teruglevering ziet (al dan niet in verschoven uren). Aan de hand daarvan is het veel makkelijker gerichter zoeken.

tojo2412, 2021-08-20:
De dagtotalen van de slimme meter zijn in ieder geval gelijk aan die van dsmr-reader. Maar ik wil graag weten wat het directe verbruik is van de zonnepanelenopbrengst op uurniveau ipv op dagniveau...

Ik heb ondertussen in mijn-fluvius het detail-verbruik aangezet (zou per kwartier zijn) om de vergelijking nogmaals te maken. Het rapport van de kwartiertotalen is pas 24u later beschikbaar, ik kom er dus later nog op terug. Bedankt in ieder geval om mee te denken !

dennissiemensma, 2021-08-20:
Ben je er nog aan toegekomen om voor **een uur** het handmatig te checken op de meter of dat wel overeen komt? Los van het (verkeerde) uur waarin DSMR-reader het nu toont.

tojo2412, 2021-08-21:
Ik was gisteren niet thuis, vandaag zal ik dat alsnog doen. 

Maar ik denk wel dat je gelijk hebt hoor. Voor de 28ste moet het 1u opgeschoven worden en na vanaf de 28ste maart 2u. Als ik het zo toepas op de export zijn er geen fouten meer mbt de data van de zonnepanelenomvormer...

tojo2412, 2021-08-22:
Hey Dennis, nieuwe update... ;-)

Als ik een export wil doen van de uurtotalen via de interface zal ik moeten wachten tot morgen. Op de één of andere manier kan hij de uurtotalen van gisteren niet vinden... 

Aangezien ik niet zoveel geduld had heb ik de API aangezet en daar de laatste 25 uurtotalen opgevraagd. Na controle worden daar wel de juiste data bij de juiste uren weergegeven.

Wil je dat ik morgen de CSV-export nog controleer ?

dennissiemensma, 2021-08-22:
Bedankt voor je update. Als de uurtotalen die je kon vinden kloppen, dan lijkt het me verder wel prima zo hoor. In de volgende versie zal de export in de lokale tijdzone zijn en ik verwacht dat het dan goed voor je werkt.

tojo2412, 2021-08-22:
De uurtotalen weken iets af (0,012 en 0,007), maar ik vermoed dat het komt omdat ik niet exact op de seconde de meterstanden heb genoteerd... 

Bedankt Dennis.

closed 2021-08-22 by tojo2412