dsmrreader/dsmr-reader #1441: Database server had een probleem, DSMR reader komt niet meer tot leven

skinkie, 2021-08-20:
Na een probleem met Postgresql (disk vol). Lukt het me niet meer om grafieken te krijgen.

[image]

Ik zie het volgende gebeuren:

[image]

[image]

Uiteraard is de schijfruimte weer beschikbaar, PostgreSQL herstart en is het uwsgi proces van dsmrreader ook herstart.

Ik heb een poging gedaan met onderstaande, maar daarmee zie ik nog geen verschil:
```bash
manage.py dsmr_stats_reconstruct_missing_day_statistics
manage.py dsmr_stats_reconstruct_missing_day_statistics_by_hours
```

dennissiemensma, 2021-08-20:
Bedankt voor je melding. Je kunt het beste niet zomaar commands uitvoeren hoor, want die zijn bedoeld voor een hele andere situatie.

DSMR-reader loopt achter met het verwerken van alle metingen en daarom zie je geen gegevens. In je screenshot zijn dat er zo'n 43.000 en als het goed is loopt die achterstand in. De exacte snelheid hangt vooral af van de hardware.

Is de achterstand inmiddels minder of helemaal weg?

skinkie, 2021-08-20:
Helaas niet.
[image]

Hardware: AMD Ryzen 5 3600 (12 cores).

dennissiemensma, 2021-08-20:
Draait het backend-proces wel? En zo ja, check dan vooral de logfile van de backend, eventueel met debug-logging ingeschakeld:
- https://dsmr-reader.readthedocs.io/nl/v4/how-to/troubleshooting/logfiles.html

skinkie, 2021-08-20:
@dennissiemensma ik denk dat dat het probleem inderdaad het backend proces was. Is het mogelijk om daar een foutmelding van te geven letterlijk "het backend proces draait niet"? Op dit moment zie ik nog geen nieuwe grafieken, maar ik heb even geduld.

skinkie, 2021-08-20:
Grafieken werken weer. Feature request blijft om aan te geven dat het backend proces niet draait.

closed 2021-08-20 by skinkie

dennissiemensma, 2021-08-20:
Fijn om te horen. Uiteindelijk denk ik dat ik meer voel voor een stappenplan om dingen te checken, want er zijn zoveel (andere) dingen waar het aan kan liggen, dan kan iemand beter een checklist afwerken wellicht.