dsmrreader/dsmr-reader #1442: Consuming en returning waardes niet correct door slimme meter (kaifa e0026).

Luukxt8, 2021-08-23:
Ik heb sinds kort zonnepanelen die terug levert op fase 1 maar heb een 3 fase slimme meter. (Kaifa E0026)

Wat me opvalt is dat de meter op het digitale schermpje wel de totale terug levering laat zien maar in de telegram het overige verbruik van fase 2 en 3 niet eraf trekt van wat ik op fase 1 terug lever. Waardoor DSMR reader dus denkt dat ik altijd stroom verbruik terwijl dit dus niet zo is.

[image]

Wat ik dat dus op het schermpje van de meter zie staan onder -P is dan 1540 (1764-224). Is er een manier waarmee ik dit kan rechttrekken? 

Alvast Bedankt.

Voorbeeld telegram:

```
/KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(210823184353S)
0-0:96.1.1(4530303236303030303137363934393135)
1-0:1.8.1(009749.901*kWh)
1-0:1.8.2(011514.222*kWh)
1-0:2.8.1(000012.720*kWh)
1-0:2.8.2(000030.750*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.301*kW)
1-0:2.7.0(01.713*kW)
0-0:96.7.21(00016)
0-0:96.7.9(00009)
1-0:99.97.0(8)(0-0:96.7.19)(210624101654S)(0000001045*s)(190602193046S)(0000001212*s)(171110020837W)(0000004279*s)(160715150757S)(0000001252*s)(160609121454S)(0000003477*s)(000101000001W)(2147483647*s)(000101000001W)(2147483647*s)(000101000001W)(2147483647*s)
1-0:32.32.0(00001)
1-0:52.32.0(00000)
1-0:72.32.0(00000)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(007*A)
1-0:51.7.0(001*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00.296*kW)
1-0:61.7.0(00.005*kW)
1-0:22.7.0(01.713*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!C724
```

Luukxt8, 2021-08-23:
#1431 Hmm balen. Net dus over deze bug heen gelezen. Hetzelfde probleem ook met een kaifa meter....

dennissiemensma, 2021-08-23:
Bedankt voor je vraag. Ik heb zelf **geen** zonnepanelen, dus ik kan alleen delen wat anderen door de jaren terug hebben gekoppeld. _Neem dit dus niet als absolute waarheid aan en informeer vooral bij een Tweakers forum bijvoorbeeld, waar veel meer gebruikers zitten met teruglevering en ook hun eigen ervaringen._

Als het goed is weergeeft de slimme meter exact zoals verbruikt en teruglevering gebeurt. Als je in jouw situatie de teruglevering niet kan verdelen over alle fasen (en ik weet niet of/hoe dat elektra-technisch uberhaupt kan tussen groepen, zelf geen ervaring mee), dan zul je zowel terugleveren als verbruiken. 
Daardoor vermoed ik dat je aanname niet klopt dat de meter het niet goed wegstreept. De meter weergeeft enkel de waarheid en heeft geen salderingsregels. Saldering gebeurt altijd door je energieleverancier en nooit door je slimme meter. Echter, zolang die salderingsregels niet veranderen en je niet over de salderingsgrens gaat (5000 kWh ofzo per jaar?), zul je er in de praktijk niets van merken. Je leverancier streept het immers voor je weg.

Mijn advies is om bovenstaande te verifieren bij iets of iemand die er 100% wat vanaf weet. Dit is puur wat ik er de afgelopen jaren van mee heb gekregen.

dennissiemensma, 2021-08-23:
Ik weet overigens niet welk forum/topic het beste is, wellicht bij een van deze?

- https://gathering.tweakers.net/forum/list_messages/1489893
- https://gathering.tweakers.net/forum/list_messages/1578510

Grote kans dat ze je daar kunnen vertellen hoe het echt werkt/hoort te werken/wat je evt nog kunt veranderen aan je opzet thuis.

dennissiemensma, 2021-08-25:
Mocht je (later) nog wat wijzer zijn geworden, of nog andere vragen hebben, laat vooral weten!

closed 2021-08-25 by dennissiemensma