dsmrreader/dsmr-reader #1445: 504 Time-out melding na upgrade naar v4

raymond79nl, 2021-08-29:
Long overdue update van v2 naar v4 gedaan vandaag. Netjes de stappen gevolgd in de documentatie: eerst naar v3 en toen naar v4. Alles zonder foutmeldingen. Alle processen draaien zoals gewenst, maar de webinterface komt niet verder dan “504 Gateway Time-out”. Die werkte wel nog bij een tussentijdse check op v3. Enig idee wat ik kan doen om dit te verhelpen? Ik zag suggesties om data logger sleep te verhogen, maar dat lijkt via interface te moeten gebeuren?

Native install op RPi 3B

dennissiemensma, 2021-08-29:
Bedankt voor je vraag. Je kunt het beste beginnen met een debug dump te delen: https://dsmr-reader.readthedocs.io/nl/latest/how-to/troubleshooting/debug-info-dump.html

raymond79nl, 2021-08-29:
Bedankt voor snelle reactie. Output hieronder:

```
DSMR-READER
    App / Python / Database                                                   v4.16.3 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                        1.0s / 2.5s / -h / False
    Latest telegram version read / Parser settings                 "50" / "4"

DATA
    Telegrams total (est.)                                                           14087300
    Consumption records electricity / gas (est.)                      596155 / 213115

UNRESOLVED ISSUES
    Too many outgoing MQTT messages queued for transit                                                    now
    Database growing large: 6355 MB, consider data cleanup (if not already enabled)        2 minutes ago
    Process behind schedule: Weather update                                                                          an hour ago
    Process behind schedule: Generate consumption data                                                      an hour ago
    Too many unprocessed readings: 3163                                                                                2 hours ago

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_mqtt_message                                                                          3612 MB
    public.dsmr_datalogger_dsmrreading                                                          2480 MB
```

dennissiemensma, 2021-08-29:
Er zijn een aantal issues die je moet oplossen, maar de grootste is denk ik het aantal opslagen metingen. Dat zijn er momenteel 14 miljoen. 
Daar is in latere DSMR-reader versies wel opschoning voor gekomen, maar zonder interface kun je die niet daar inschakelen, en de interface is vermoedelijk pas weer beschikbaar als de hoeveelheid data wat binnen de perken valt, dus zul je dat via de database moeten doen. 

Je kunt de opschoning bijvoorbeeld op een maand zetten:
```
sudo su - postgres
psql dsmrreader

update dsmr_datalogger_retentionsettings set data_retention_in_hours = 672;
```

Als het goed is zie je de wijziging dan ook terug in een nieuwe debug dump op de regel:
```
    BE sleep / DL sleep / Retention / Override                        1.0s / 2.5s / -h / False
```

Er kan overigens wel zo een paar dagen overheen gaan voordat alles opgeschoond is, maar zolang DSMR-reader nieuwe metingen blijft registreren is dat datatechnisch prima.

----

Een ander dingetje is dat er heel veel MQTT-bericht gequeued zijn. Ik weet niet of je MQTT wel of niet gebruikt, maar als je die niet gebruikt, kun je het beste die tabel leeggooien en later MQTT uitschakelen in DSMR-reader.

```
sudo su - postgres
psql dsmrreader

delete from dsmr_mqtt_message;
```

dennissiemensma, 2021-08-29:
Overigens zul je die tellers in de debug dump niet direct naar beneden zien gaan. Het zijn geen exacte aantallen omdat dat bij grote tabellen anders te lang zou duren (minpuntje van PostgreSQL).

Je zou wel af en toe handmatig een count kunnen doen. En **na een tijdje** eenmalig een `vacuumdb` (geeft ook de schijfruimte weer terug). Voor nog wat meer informatie zie: #1208

```
sudo su - postgres
vacuumdb -f -v -z -d dsmrreader
```

raymond79nl, 2021-08-30:
Top, dat (database opschoning ingesteld + verwijdering MQTT berichten) heeft in ieder geval geholpen! Ik kan nu in de webinterface, die normaal reageert maar geen dag statistieken laat zien hoewel het live verbruik wel updated wordt bovenaan de main page. Nu ik in de web interface kan: MQTT lijkt ook uit te staan: in “MQTT broker/verbinding wijzigen” staat geen vinkje bij “ingeschakeld”. 

Nieuwe debug dump:

```
DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                    1.0s / 10.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                           14087300
    Consumption records electricity / gas (est.)                                              596155 / 213115

UNRESOLVED ISSUES
    Database growing large: 6357 MB, consider data cleanup (if not already enabled)                                     now
    Process behind schedule: Generate day and hour statistics                                  19 minutes ago
    Process behind schedule: Weather update                                                    19 minutes ago
    Process behind schedule: Generate consumption data                                         24 minutes ago
    Too many unprocessed readings: 3492                                                           9 hours ago
    Day statistics are lagging behind                                                      1 day, 8 hours ago

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_mqtt_message                                                                          3612 MB
    public.dsmr_datalogger_dsmrreading                                                                2483 MB
```

dennissiemensma, 2021-08-30:
Fijn om te horen! Alle achterstanden loopt die vanzelf na een tijd in. Nu is het vooral wachten tot de opschoning klaar is. Dat kan enige tijd duren, maar gaat wel steeds sneller.

Je kunt iig al de MQTT-message tabel "opschonen":

```
sudo su - postgres
vacuumdb -t public.dsmr_mqtt_message -f -v dsmrreader
```

Als het goed is verdwijnt de melding over die tabel in de debugdump/interface.

----

Daarna kun je af en toe de `dsmr_datalogger_dsmrreading` laten analyseren (geen idee hoe snel dat gaat).

```
sudo su - postgres
vacuumdb --analyze-only -t dsmr_datalogger_dsmrreading -v dsmrreader
```

Daarmee zou deze waarde opnieuw berekend moeten worden in de debug dump:
```
    Telegrams total (est.)                                                                           14087300
```

----

Je kunt ook altijd een directe record-count doen in die tabel, maar opzich is het allemaal puur ter info. En als die helemaal bij is, verdwijnen alle meldingen behalve deze:

```
    public.dsmr_datalogger_dsmrreading                                                                2483 MB
```

In dat geval kun je tot slot dan deze doen:
```
vacuumdb -f -v -z -d dsmrreader
```

Vanaf dit punt zal je DB nooit meer zo snel groeien, omdat de opschoning hem altijd bijhoudt. En de performance van DSMR-reader zou vanaf dan ook weer prima moeten zijn.

dennissiemensma, 2021-09-01:
Is je achterstand inmiddels weg en heeft bovenstaande de waarschuwingen in DSMR-reader verholpen?

raymond79nl, 2021-09-02:
Dank voor de opvolging. Ik had hem een paar dagen laten lopen zonder naar de meldingen te kijken. Vanmorgen naar de meldingen in de web interface gekeken en toen was inderdaad alleen de melding over DB grootte over. Na het draaien van het `vacuumdb -f -v -z -d dsmrreader` commando is ook die weg en draait dus alles als een zonnetje. Super bedankt voor de ondersteuning!

dennissiemensma, 2021-09-03:
Top, fijn om te horen!

closed 2021-09-03 by dennissiemensma