dsmrreader/dsmr-reader #1446: Archief verbruik per week inzien

niekniek89, 2021-09-02:
Goedemiddag,

Een vraag (en geen idee of dit gewenst/mogelijk is):
Momenteel kan je het archief gebruiken voor dag/maand/jaarverbruik.
Is het ook mogelijk om het weekverbruik in te zien in het archief?

Het zou wellicht een "leuke" toevoeging zijn, maar zoals hierboven.. geen idee of dit gewenst/mogelijk is.

dennissiemensma, 2021-09-03:
Bedankt voor je verzoek. Ik zal het op de lijst van _mogelijke_ toekomstige features zetten. Het is namelijk helaas niet heel triviaal om in te bouwen, gezien de gebruikte datepicker geen weekselectie ondersteunt.

dennissiemensma, 2021-09-22:
Het is behoorlijk wat werk en ik denk dat het het niet waard is, tov andere tools die mogelijk hetzelfde (en allerlei soorten intervals) makkelijker kunnen bieden. Meer toelichting in soortgelijk verzoek: https://github.com/dsmrreader/dsmr-reader/issues/1348#issuecomment-923339346

closed 2021-09-22 by dennissiemensma