dsmrreader/dsmr-reader #1447: Verkeerde huidig verbruikswaarde op dashboard

ThaStealth, 2021-09-04:
Hi,

Ik gebruik de API om data in DSMR Reader te krijgen, ik gebrijk DSMR-reader v4.16.3. Als ik de volgende JSON post:
```
{
    "timestamp": "2021-09-04T19:31:40Z",
    "electricity_delivered_1": 5974.408,
    "electricity_returned_1": 2864.873,
    "electricity_delivered_2": 2737.308,
    "electricity_returned_2": 6532.458,
    "electricity_currently_delivered": 563,
    "electricity_currently_returned": 0,
    "phase_currently_delivered_l1": 116,
    "phase_currently_delivered_l2": 98,
    "phase_currently_delivered_l3": 347,
    "phase_currently_returned_l1": 0,
    "phase_currently_returned_l2": 0,
    "phase_currently_returned_l3": 0,
    "phase_voltage_l1": 231,
    "phase_voltage_l2": 232,
    "phase_voltage_l3": 232,
    "phase_power_current_l1": 0,
    "phase_power_current_l2": 0,
    "phase_power_current_l3": 1
}
```
(de documentatie zegt overigens dat er strings verwacht worden, maar volgens mij worden er gewoon floats verwacht):
```
{
"timestamp": "2019-08-24T14:15:22Z",
"electricity_delivered_1": "string",
"electricity_returned_1": "string",
"electricity_delivered_2": "string",
"electricity_returned_2": "string",
"electricity_currently_delivered": "string",
"electricity_currently_returned": "string",
"phase_currently_delivered_l1": "string",
"phase_currently_delivered_l2": "string",
"phase_currently_delivered_l3": "string",
"extra_device_timestamp": "2019-08-24T14:15:22Z",
"extra_device_delivered": "string",
"phase_currently_returned_l1": "string",
"phase_currently_returned_l2": "string",
"phase_currently_returned_l3": "string",
"phase_voltage_l1": "string",
"phase_voltage_l2": "string",
"phase_voltage_l3": "string",
"phase_power_current_l1": 0,
"phase_power_current_l2": 0,
"phase_power_current_l3": 0
}
```

Dan krijg ik deze reactie terug:
```
{
    "id": 62313337,
    "timestamp": "2021-09-04T21:31:40+02:00",
    "electricity_delivered_1": "5974.408",
    "electricity_returned_1": "2864.873",
    "electricity_delivered_2": "2737.308",
    "electricity_returned_2": "6532.458",
    "electricity_currently_delivered": "563.000",
    "electricity_currently_returned": "0.000",
    "phase_currently_delivered_l1": "116.000",
    "phase_currently_delivered_l2": "98.000",
    "phase_currently_delivered_l3": "347.000",
    "extra_device_timestamp": null,
    "extra_device_delivered": null,
    "phase_currently_returned_l1": "0.000",
    "phase_currently_returned_l2": "0.000",
    "phase_currently_returned_l3": "0.000",
    "phase_voltage_l1": "231.0",
    "phase_voltage_l2": "232.0",
    "phase_voltage_l3": "232.0",
    "phase_power_current_l1": 0,
    "phase_power_current_l2": 0,
    "phase_power_current_l3": 1
}
```

Echter krijg ik op het dashboard 563000 te zien als huidig verbruik, ik denk dat je geen rekening houd met de decimal seperator. en de waarde dus x1000 doet. Bij de andere waardes zie ik dit probleem niet terug
Voorbeeldje (is van een ander bericht, maar het effect is hetzelfde): 
[image]

dennissiemensma, 2021-09-04:
Bedankt voor je melding. De getallen die je moet doorgeven zijn in kWh en kW. Daarom wordt het ook daadwerkelijk als x 1000 verrekend en getoond. 
Hoewel er ook wel wat te zeggen was voor bijvoorbeeld Watt of Watt/uur, heb ik het maar gelijk gehouden met wat de meters teruggeven.

Het staat ook in de docs, maar ik kan me voorstellen dat er zonder voorbeeld-request en zonder voorbeeld-data snel overheen gelezen wordt:
[image]

----

Tot slot, qua strings/floats is dit iets wat het API-framework handhaaft vermoed ik, zie ook #1319 en ik vermoed dat het voor input ook zo is, al pakt die vast ook gewoon JSON numbers.

----

Ik vermoed dat je request zoiets mag worden:

```
{
    "timestamp": "2021-09-04T19:31:40Z",
    "electricity_delivered_1": 5974.408,
    "electricity_returned_1": 2864.873,
    "electricity_delivered_2": 2737.308,
    "electricity_returned_2": 6532.458,
    "electricity_currently_delivered": 0.563,
    "electricity_currently_returned": 0,
    "phase_currently_delivered_l1": 0.116,
    "phase_currently_delivered_l2": 0.98,
    "phase_currently_delivered_l3": 0.347,
    "phase_currently_returned_l1": 0,
    "phase_currently_returned_l2": 0,
    "phase_currently_returned_l3": 0,
    "phase_voltage_l1": 231,
    "phase_voltage_l2": 232,
    "phase_voltage_l3": 232,
    "phase_power_current_l1": 0,
    "phase_power_current_l2": 0,
    "phase_power_current_l3": 1
}
```

dennissiemensma, 2021-09-06:
Mocht je nog tegen andere issues aanlopen, laat maar weten!

closed 2021-09-06 by dennissiemensma