dsmrreader/dsmr-reader #1448: Te veel onverwerkte telegrammen: 48431

reneposthumus, 2021-09-07:
Hi Dennis,

Sinds een aantal dagen loopt de verwerking achter, en ik heb andere issues die hiermee te maken hebben ook gelezen maar de MQTT verbinding is verder prima, ook geen errors in de debuglogs van de supervisor. Na verder gezocht te hebben in de logs heb ik die van postgresql bekeken, hier staat wel steevast een error in waarvan ik vermoed dat er iets mis mee is:

`**2021-09-07 09:27:01.020 BST [6438] dsmrreader@dsmrreader ERROR:  index "dsmr_consumption_electrici_currently_delivered_0f28e53a" contains unexpected zero page at block 42**
2021-09-07 09:27:01.020 BST [6438] dsmrreader@dsmrreader HINT:  Please REINDEX it.
2021-09-07 09:27:01.020 BST [6438] dsmrreader@dsmrreader STATEMENT:  INSERT INTO "dsmr_consumption_electricityconsumption" ("read_at", "delivered_1", "returned_1", "delivered_2", "returned_2", "currently_delivered", "currently_returned", "phase_currently_delivered_l1", "phase_currently_delivered_l2", "phase_currently_delivered_l3", "phase_currently_returned_l1", "phase_currently_returned_l2", "phase_currently_returned_l3", "phase_voltage_l1", "phase_voltage_l2", "phase_voltage_l3", "phase_power_current_l1", "phase_power_current_l2", "phase_power_current_l3") VALUES ('2021-09-01T16:47:00+00:00'::timestamptz, '12365.189', '4344.305', '12056.152', '10108.866', '0.340', '0.000', NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL) RETURNING "dsmr_consumption_electricityconsumption"."id"`

[image]

[image]

Kun je een oordeel vellen hierover? En me een duw in de oplossingsrichting geven?

Alvast dank!

René

dennissiemensma, 2021-09-07:
Bedankt voor je melding. Ik denk dat je als **allereerste** zo snel mogelijk een backup van de DB moet maken en/of door DSMR-reader gemaakte DB-backups (in `/home/dsmr/dsmr-reader/backups/` veiligstellen **buiten** je systeem. Helemaal als je dit op een Pi met SD-kaart draait, want in dat geval is de kans groot dat er beginnende corruptie is.

Daarna kun je kijken of de reindex uit de hint helpt:
```
sudo su - postgres

psql dsmrreader
reindex index dsmr_consumption_electrici_currently_delivered_0f28e53a;
```

Ik weet niet of je daarna DSMR-reader opnieuw moet herstarten, maar je kunt dat altijd proberen.
```
sudo supervisorctl restart all
```

Let wel dat het moeilijk te zien is of je kaartje (als je een SD-kaart gebruikt) corrupt is, of dat dit een andere oorzaak heeft. In het ergste geval zul je een herinstallatie moeten doen en een back-up terugzetten. Vandaar dat laatstgenoemde het belangrijkste is om als eerste veilig te stellen.

reneposthumus, 2021-09-08:
> Bedankt voor je melding. Ik denk dat je als **allereerste** zo snel mogelijk een backup van de DB moet maken en/of door DSMR-reader gemaakte DB-backups (in `/home/dsmr/dsmr-reader/backups/` [...]

Hi Dennis,

Dank voor je snelle reactie. Aangezien ik automatisch met SCP backups maak had ik nog een backup welke niet op de Pi zelf staat.

Ik draai het dus inderdaad op een Pi, maar met Berryboot zodat ik iSCSI opslag op een Synology kan gebruiken. Het wil wel eens gebeuren dat de verbinding even onderbroken wordt dus corruptie in de hand kan werken.

Ik heb per jouw advies een reindex gedraaid, maar had niet het resultaat waar ik op gehoopt had. 

Wat ik uiteindelijk geprobeerd heb, is de database weggooien, nieuwe database maken en een backup terugzetten van voordat de ellende begon middels jouw documentatie hierover. En dat werkt!

Er begint al een en ander weer zichtbaar te worden:

[image]

[image]

bikeymouse, 2021-09-08:
Hi,
Ik heb ook sinds kort dit soort meldingen.  
Het verdwijnt na het herstarten van de DSMR container (ik gebruik de xirixiz Docker versie). Ik had het idee dat het te maken heeft met het hangen van een proces. Dacht dat het na een aantal herstarts was opgelost, maar zie dat het net weer is opgetreden.

Ik krijg nu het vermoeden dat het te maken heeft met het ophalen van de weergegevens via de Buienradar API, want de timing is hetzelfde en toen ik dat uit had gezet draaide het 2 dagen zonder problemen. Nu overigens 6 dagen zonder fouten, maar nu ik kijk, dus toevallig net een uur geleden weer gestopt met verwerken van telegrammen:

[image]

```

DSMR-READER
    App / Python / Database                                                     v4.16.3 / v3.9.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                   15.0s / 15.0s / 168h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                              83475
    Consumption records electricity / gas (est.)                                                84673 / 42882

UNRESOLVED ISSUES
    Te veel onverwerkte telegrammen: 283                                                      een uur geleden
    Proces loopt achter op schema: Generate consumption data                                  een uur geleden

```
De enige recente fout die ik kan vinden is in dsmr_backend.log maar komt denk ik door het herstarten van de container. Ik heb geen fouten in de postgress database container logs.

```
2021-09-02 17:24:54,924 ERROR    mixins       run_once                        101 | dsmr_backend.management.commands.dsmr_backend: [!] Exception raised. Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.AdminShutdown: terminating connection due to administrator command
server closed the connection unexpectedly
        This probably means the server terminated abnormally
        before or while processing the request.
```

dennissiemensma, 2021-09-08:
> Het enige dat nog over blijft zijn de dagstatistieken. Lost dat zichzelf op?

@reneposthumus afhankelijk van hoe oud je backup was, verwacht ik dat die ofwel de achterstand vanzelf inloopt of dat de melding na aankomende nacht weg gaat.

dennissiemensma, 2021-09-08:
@bikeymouse bedankt voor je melding en hint. Ik zal in ieder geval een timeout toevoegen aan de request naar Buienradar, mocht die (soms) daarop hangen.

closed 2021-09-08 by dennissiemensma

bikeymouse, 2021-09-08:
> @bikeymouse bedankt voor je melding en hint. Ik zal in ieder geval een timeout toevoegen aan de request naar Buienradar, mocht die (soms) daarop hangen.

Ok dank! Zou het geen antwoord krijgen van de Buienradar API inderdaad bovenstaand gedrag kunnen verklaren?

dennissiemensma, 2021-09-08:
> Ok dank! Zou het geen antwoord krijgen van de Buienradar API inderdaad bovenstaand gedrag kunnen verklaren?
Ja dat zou kunnen, alleen kan ik het niet makkelijk verifieren/reproduceren. 

Als het al op die manier fout gaat, dan zou je na een tijdje van een hoop processen in DSMR-reader een waarschuwing krijgen dat ze achterlopen. Dat beperkt zich dan als het goed is niet tot alleen specifiek telegramverwerking.

En ik zou ook fouten verwachten in de logfile van de backend. Ook zonder debug-logging aan.

bikeymouse, 2021-09-08:
Ik had nu ook dat “Generate consumption data” achterliep. En toen ik het eerder had en het langer had geduurd had, had ik dacht ik ook meldingen dat de daystatistics achterliepen. Maar geen fouten in de backend-log.

Ik heb in ieder geval de weer-update optie nu uitgezet en zal komende dagen in de gaten houden of ik nieuwe foutmeldingen krijg.

Dank weer voor de snelle reactie!