dsmrreader/dsmr-reader #1449: Oude meterstanden

cmderks93, 2021-09-09:
Zou het een mogelijkheid kunnen zijn om een serie oude meterstanden in te voeren in de database?
Ik heb een aantal meterstanden in een csv bestand staan namelijk (enelogic) maar zou het fijn vinden als dit toch bij de DSMR gevoegd kan worden.

dennissiemensma, 2021-09-10:
Bedankt voor je vraag. Dit is zeker mogelijk, maar het hangt af van welke gegevens beschikbaar zijn. Heb je een voorbeeld van de gegevens?

cmderks93, 2021-09-10:
[image]
Dit het meest actuele wat ik kan vinden, dus een beetje info die ook uit de telegram komt. Geen vermelde tijden (is de 00:00 uur stand zover ik kan vinden) Ik kan ook het verbruik per kwartier inzien, maar dit is al berekend, en dus geen standen. Het blijkt dat de kwartierstanden en uurstanden niet doorgegeven worden, maar alleen het verbruik.

dennissiemensma, 2021-09-10:
Bedankt voor je aanvulling. Ik denk dat je een heel eind kunt komen als je de meterstanden in een Excelsheet/CSV krijgt en het verbruik per type per dag ermee uitrekent. 
Als je zowel het verbruik als de meterstanden per dag op 1 regel met verschillende kolommen krijgt, kan ik wel voor je kijken om het om te zetten naar een bulk aan queries om ze als dagtotalen toe te voegen. Zeg maar de gegevens op: `/admin/dsmr_stats/daystatistics/add/`

cmderks93, 2021-09-10:
[image]
Dit heb ik van onze leverancier, daar kunnen we wel wat mee denk ik.

dennissiemensma, 2021-09-10:
Uiteindelijk worden alle gegevens in DSMR-reader verwerkt als dagstatistieken. Die zie je ook in het Archief en worden ook gebruikt bij de pagina voor de Energiecontracten (als je dat ingesteld hebt). Bij de dagstatistieken kun je wel of geen meterstanden opnemen, net wat je wil.
Verder zijn er in DSMR-reader ook nog uurstatistieken, die geven puur het verbruik aan en worden gebruikt bij de dagweergave in het Archief. En voor Trends dacht ik.

Ik denk dat je eerste screenshot een goed startpunt is, maar dan wordt de uitdaging voor je om het op 1 regel te krijgen per dag. Overigens moeten de meterstanden wel in kWh zijn, want ik kan niet helemaal afleiden uit die waardes of kloppen. Of je moet op 30 augustus ~`261 kWh` hebben verbruikt... :smile:

cmderks93, 2021-09-10:
Heb even de pagina bekeken, en zie idd wat je bedoelt. Ik heb 2 databronnen :-) enelogic, en onze leverancier. 
Ik heb de meterstanden per dag, en de verbruikscijfers per uur. Zou het mooiste zijn als we ze beide kunnen verwerken. 
Zou ik dit in 2 bestanden kunnen aanleveren, dat het in 2 imports verwerkt zal worden?
Verbruik zal wel kloppen, is een restaurant :-)

dennissiemensma, 2021-09-11:
Je zult eerst zelf eea moeten berekenen zodat er een Excel/CSV uit komt met onderstaande formaten. Als je het in een uniform formaat werkend hebt kunnen we het wel uitwisselen en kan ik het voor je omzetten naar DB-queries. 

## Dagstatistieken
Dit is een voorbeeld, het gaat vooral om welke gegevens op 1 regel horen en welke van deze velden noodzakelijk zijn.
```
  id  |    day     | total_cost | electricity1 | electricity2 | electricity1_returned | electricity2_returned | electricity1_cost | electricity2_cost |  gas  | gas_cost | average_temperature | highest_temperature | lowest_temperature | fixed_cost | electricity1_reading | electricity1_returned_reading | electricity2_reading | electricity2_returned_reading | gas_reading 
------+------------+------------+--------------+--------------+-----------------------+-----------------------+-------------------+-------------------+-------+----------+---------------------+---------------------+--------------------+------------+----------------------+-------------------------------+----------------------+-------------------------------+-------------
 2350 | 2021-09-10 |       2.14 |        1.246 |        5.258 |                 0.000 |                 0.000 |              0.22 |              1.02 | 0.429 |     0.30 |                19.3 |                25.5 |               15.7 |       0.60 |             5487.388 |                         0.000 |             5641.312 |                         0.000 |    4046.255
```

- `day` (Y-m-d)
- `electricity1` (kWh verbruik)
- `electricity2` (kWh verbruik)
- `electricity1_returned` (kWh teruglevering)
- `electricity2_returned` (kWh teruglevering)
- `gas` (m3 verbruik)
- `electricity1_reading` (eerste laagrarief meterstand vd dag in kWh)
- `electricity2_reading` (eerste hoogtarief meterstand vd dag in kWh)
- `electricity1_returned_reading` (eerste laagrarief meterstand vd dag in kWh)
- `electricity2_returned_reading` (eerste hoogtarief meterstand vd dag in kWh)
- `gas_reading` (eerste gasmeterstand vd dag in m3)

## Uurstatistieken
```
  id   |       hour_start       | electricity1 | electricity2 | electricity1_returned | electricity2_returned |  gas  
-------+------------------------+--------------+--------------+-----------------------+-----------------------+-------
 52536 | 2021-09-10 22:00:00+01 |        0.171 |        0.003 |                 0.000 |                 0.000 | 0.000
```

- Alle velden behalve `id`
- Let op het formaat van `hour_start` met de tijdzone in wintertijd/zomertijd

----

Ik zou beginnen met de dagstatistieken, gezien je daar het meeste aan hebt. Als je zelf handig bent met scripten, kun je het uiteraard ook zelf doen of via de DSMR-reader API. En anders maak ik er wel gewoon queries voor je van.

cmderks93, 2021-09-11:
Top!
Ga ik me eens mee bezig houden:-)

Thanks voor je hulp en uitleg!

dennissiemensma, 2021-09-11:
Ter aanvulling, je kunt ook nog kijken naar `/export` in DSMR-reader, als je daar een dump maakt van je dag- of uurstatistieken. Zo'n formaat is ook prima. 
_Let wel dat voor de uurstatistieken de tijdzone daar momenteel niet klopt (fix in volgende release)._

cmderks93, 2021-09-17:
Ik heb een aantal van de verplichte velden niet in gebruik, terugleveren en gas. 
Deze kunnen leeg blijven? Of moet hier wel een 0 ingevuld worden oid?

dennissiemensma, 2021-09-18:
Voor teruglevering kun je inderdaad `0` gebruiken. Voor gas mag die leeg (`null`) zijn, maar als dat lastig is kan die eventueel op op `0` en dan kun je die achteraf wel op `null` zetten of simpelweg gasverbruik in DSMR-reader "onderdrukken".

cmderks93, 2021-09-24:
Nou, ik heb eindelijk wat :-)
[image]
Is dit zo juist?

dennissiemensma, 2021-09-26:
Ja ik denk dat dit al goed genoeg is voor een eerste poging. Als je wilt kunt je proberen er zelf insert-queries van te maken, maar ik kan ook wel een poging voor je doen. 
Ik kan ze dan hier in mijn ontwikkelomgeving even proberen totdat ze goed importeren en dan stuur ik je de INSERT-queries terug die je zelf redelijk makkelijk kan uitvoeren op je installatie. Voor dat laatste mag je de data/sheet evt naar me mailen zodat je het niet hier publiekelijk hoeft te delen. Dat kan naar github at dennissiemensma . nl

cmderks93, 2021-09-27:
Ik heb het je gemaild. Dank voor je hulp!

cmderks93, 2021-09-28:
Dankje! voor het aanmaken van de SQL query. Het is gelukt!

dennissiemensma, 2021-09-28:
Aanvullend: Er komt in 4.18 nog een fix voor lege gaswaardes in `dsmr_stats_recalculate_prices`

closed 2021-09-28 by dennissiemensma