dsmrreader/dsmr-reader #1450: dsmr sensoren vernieuwen niet in Home Assistant

Jos-77, 2021-09-15:
Ik heb de dsmr_reader add on geinstalleerd volgens de handleiding, in de web UI zie ik dat er gegevens binnenkomen, helaas worden deze niet in home assistant getoond het is net of de boel niet vernieuwd wordt.

Het vreemde is dat de toestand van 2 dagen geleden blijft staan als de waarde, ook al wordt er een herstart gedaan.

Datalogger draait op een rpi 3b+ waar ook unifi op draait, ook al geprobeert met alleen de datalogger, geen verschil, de waarden worden verzonden naar mijn HA. HA draait op i7 systeem in Oracle Virtualbox, alles draait als een zonnetje behalve de dsmr-reader add on.  graag een oplossing.

Mocht er meer info nodig zijn, dan geef ik deze graag.

Mijn configuration.yaml

sensor:
# DSMR Reader
  - platform: dsmr_reader
    host: 192.168.2.45
    port: 7777
    dsmr_version: 4

heb het ook al geprobeerd met versie 4.2

Entiteit
sensor.dsmr_day_consumption_electricity2
 
Toestand
1.604
Status attributen (YAML, optioneel)

state_class: total_increasing
unit_of_measurement: kWh
friendly_name: High tariff usage
device_class: energy

Laatst gewijzigd:
15 september 2021 11:43:01

Laatst bijgewerkt:
15 september 2021 11:43:01

Jos-77, 2021-09-15:
net mijn hele database eruit gegooid, en de enige gegevens, die met dsmr-reader te maken hebben, die in het logboek van MariaDB komen zijn:

Rejected telegrams gewijzigd in 3
12:01:45 - 7 minuten geleden
Voltage swell L1 gewijzigd in 0
12:01:45 - 7 minuten geleden
Voltage sag L1 gewijzigd in 0
12:01:45 - 7 minuten geleden
Long power failure count gewijzigd in 3
12:01:45 - 6 minuten geleden
Power failure count gewijzigd in 5
12:01:45 - 6 minuten geleden
Electricity tariff gewijzigd in high
12:01:45 - 6 minuten geleden
DSMR version gewijzigd in 4.2
12:01:45 - 6 minuten geleden

dennissiemensma, 2021-09-15:
Bedankt voor je melding. Dit project is specifiek voor DSMR-reader, de reader met (een) visualizatie van de gegevens. Helemaal als DSMR-reader zelf qua interface goed lijkt te werken, houdt het voor mij snel op qua hulp kunnen bieden bij debuggen van HA. 

Voor issues met de HA addon, kun je het beste naar de maintainer gaan. Ik weet niet precies welke je gebruikt, maar wellicht deze? 
- https://community.home-assistant.io/t/dsmr-reader-add-on-for-home-assistant/279087
- https://github.com/sanderdw/hassio-addons/issues

Je kunt hooguit nog kijken of DSMR-reader de MQTT-berichten (er vanuitgaande dat het via MQTT gaat) goed en regelmatig verstuurt naar je broker: #1433

Jos-77, 2021-09-16:
overgezet naar de juiste github.

closed 2021-09-16 by Jos-77