dsmrreader/dsmr-reader #1452: Set a separate frequency for full backup on DropBox and/or re-use same file name for all backups

JeanMiK, 2021-09-18:
I am running two instances of dsmrreader in docker containers on a raspberry pi, and have them do daily backups. I also activated the backup to a common DropBox account. But with the size of the dbs growing over time, the combined size of the backups exceeded the space available on the free version of DropBox.

This is mainly caused by the seven distinct backup copies, one for each workday. As the system is already doing daily partial backups, it could be useful to reduce the frequency of the full backps to DropBox, so that e.g. only one copy is done per week, instead of 7.

Also, as DropBox offers already a version history of files, it would actually not be needed to have different names for the different copies in DropBox. If DSMRREADER was using always the same name for every backup, the user could have 30 days of backups while only being charged for the size of the current version.

dennissiemensma, 2021-09-18:
Thanks for your suggestion. The current rotation of backups was not meant for Dropbox, as Dropbox was more of a failsafe. The rotation is to make sure backups are not overwritten daily, as not every user uses Dropbox at all. 
There were some unfortunate situations in the past where it could have saved data against corruption, even though nowadays I think it may be less of an issue. 

The current way of backing up and Dropbox sync may change in the future anyway, since DSMR-reader should only provide or document an easy way of dumping and restoring, leaving the actual backup storage/sync to other tools/services. This remove drop the (unintentional) Dropbox vendor lock and its maintainance in DSMR-reader, at the expense of some additional steps for users after installation. It's not planned soon though, probably not before DSMR-reader v6.x.

Nonetheless I think you have two options at this time:
- Either reduce the data size by using retention in DSMR-reader.
- Or (I presume you do not use the latter by choice as datasize would not be an issue at all), disable backups in DSMR-reader entirely and manually schedule a DB backup using a static name in the same folder as the DSMR-reader backups.

E.g. have the daily cron run:
```
sudo -u postgres pg_dump -d dsmrreader | gzip --fast > /home/dsmr/dsmr-reader/backups/dsmrreader-manual-postgresql-backup.sql.gz
```

_Please note as a side note that not reducing the data in DSMR-reader **may** cause performance issues at some point, depending on the hardware and amount of data._

closed 2021-09-19 by dennissiemensma

dennissiemensma, 2022-11-21:
An undocumented envvar was added later in #1685, but will probably replaced later by a built-in one.