dsmrreader/dsmr-reader #1453: Gemiddelden gas in trends verdwijnen door kleine waarden

FredericMa, 2021-09-27:
Ik gebruik DSMR-reader sinds 23 juni.
Gezien wij zonneboiler hebben, hebben we sindsdien weinig gas verbruikt.
Het is me opgevallen dat op gegeven momenten bijvoorbeeld 6 stukken op de piechart zichtbaar zijn en een paar dagen/weken erna plotseling maar 4. Zie hieronder:
Van 23 juni tot 11 september:
[image]
Van 23 juni tot 26 september:
[image]

Ik heb gemerkt dat de gemiddelden die via de API worden doorgestuurd een kleine waarde hebben nl 0.02 en 0.01.
We hebben in totaal 6.178 m³ gas verbruikt in de totale periode van ongeveer 100 dagen.
Als ik het goed heb, worden bij de berekening van de gemiddelden ook de uren met een verbruik van 0 m³ meegeteld. Dit zorgt ervoor dat het gemiddelde naar beneden getrokken wordt waardoor het na verloop van tijd onder 0.01 duikt en dus op 0 valt waardoor het ook verdwijnt uit de grafiek.
Als oplossing kan je de uren met waarde 0 negeren bij het berekenen van het gemiddelde. Anderzijds kan het ook een oplossing zijn om het aantal decimalen te verhogen zodat de waarde in de API niet op 0 valt.

Laat me gerust weten als deze uitleg niet duidelijk is of als je meer gegevens nodig hebt.

FredericMa, 2021-09-28:
Thanks! I'll test later today.

FredericMa, 2021-09-28:
Ik heb het getest en het geeft een veel beter resultaat. Bedankt!
[image]

Is het een idee om de records met een verbruik of teruglevering van 0 eruit te filteren voor de berekening van de trends?
Dit zou een nog acuurater resultaat geven met veel kleinere afrondingsfouten.

dennissiemensma, 2021-09-28:
Fijn om te horen!

Ik weet niet of het veel uit gaat maken als ik 0 eruit filter, omdat het effectief het gemiddelde (percentage) in alle uren evenredig aantast. Als het absolute gemiddelde in kWh/m3 weergegeven zou worden, zou het wel noodzakelijk zijn denk ik.

dennissiemensma, 2021-09-29:
Uitgebracht in [v4.18](https://github.com/dsmrreader/dsmr-reader/discussions/1456)