dsmrreader/dsmr-reader #1454: probeer te updaten

ssewalt, 2021-09-28:
ik probeer te updaten van 4.16.3 naar 4.17.1.
Gewoon via ./deploy.sh

Wat ik ook probeer ik blijf melding krijgen 

pi@DSMR:~ $ sudo su - dsmr
(dsmrreader) dsmr@DSMR:~/dsmr-reader $ ./deploy.sh

 --- You are currently running version:
4.16.3

 --- Checking Python version.
Running Python 3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0]

 --- Checking for local file changes.
 M backups/.gitkeep
 M backups/archive/.gitkeep
 M backups/archive/README.txt

[!] Local file changes detected, aborting deployment
    To revert all files to their default, WITHOUT backup, execute:

    git reset --hard HEAD

Geen idee hoe op te lossen.

dennissiemensma, 2021-09-28:
Bedankt voor je melding. Je kunt het commando onderaan de melding uitvoeren en daarna nogmaals een deploy proberen.

ssewalt, 2021-09-29:
Dennis, dank voor je terugkoppeling.  De git-reset uitgevoerd en daarna nogmaals de deploy geprobeerd; maar helaas blijf dezelfde melding krijgen. Dit vaker uitgevoerd; maar het nummer van de HEAD blijft gelijk.  Wat ik verder ook geprobeerd heb is dsmr reader uitgezet en dan de deploy. maar melding blijft gelijk.

dennissiemensma, 2021-09-29:
Wat is de output van `git reset --hard HEAD`?

Je kunt anders nog proberen de bestanden te verwijderen en daarna de reset te doen, zodat ze weer teruggezet worden. Het zijn allemaal placeholders of onbelangrijke bestanden voor de inhoudelijke werking.

```
sudo su - dsmr
rm backups/.gitkeep backups/archive/.gitkeep backups/archive/README.txt
git reset --hard HEAD
```

closed 2021-09-29 by dennissiemensma