dsmrreader/dsmr-reader #1493: Feature request: energiecontract invoer losse vaste kosten gas/stroom

mdr-github, 2021-10-04:
Hallo Dennis, allereerst mijn complimenten voor het fantastische DSMR! Ik gebruik het nu een half jaar met een custom P1 datalogger (ESP8266). DSMR draait in een docker container op mijn OpenMediaVault thuisserver, een top combinatie wat mij betreft.

Ik had eerst alleen een slimme stroom meter maar na een meterwissel krijg ik vandaag voor het eerst ook gas gegevens binnen. Daarom wilde ik de prijsgegevens van mijn gascontract invoeren. Ik heb twee losse leveranciers voor stroom en gas, met contracten die niet overlappen. Geen probleem dacht ik want ik kan die separaat invoeren, ongebruikte prijzen zet ik dan op 0.

Het probleem ontstaat bij het invoeren van de vaste kosten. Daar is één invoer "Fixed daily costs" per contract. Maar wanneer ik die probeer in te voeren merkt het systeem terecht op dat er al een invoer bestaat voor dat veld in een ander contract.

Is het mogelijk de Fixed Costs invloer te splitsen in Fixed Costs Electricity en Fixed Costs Gas? Dan zou de juiste vaste kosten bij het juiste verbruik kunnen worden opgeteld. Alvast dank.

Hartelijke groet,
Marc

dennissiemensma, 2021-10-04:
Bedankt voor je vriendelijke woorden en je verzoek. De prijzen in DSMR-reader zijn van origine bedoeld als indicatie wat een dag ongeveer kost qua verbruik.  In principe wordt het prijssysteem van DSMR-reader niet uitgebreid wegens bijbehorende complexiteit en de ontelbare combi's qua prijzen, kortingen, vaste kosten, etc. 

Ik kan de vaste kosten sowieso niet splitsen omwege bovenstaande, maar wellicht dat ik dit in de toekomst op een andere manier doe, zodat overlappende contracten vanaf dan bij elkaar opgeteld kunnen worden.

Echter is dat geen triviale wijziging en niet op korte termijn, dus de enige workaround die er nu voor je is is om de vaste kosten bij elkaar op te tellen en onder 1 contract te zetten (1 van beide). 
Wellicht dat je er zelfs een los contract voor kan invoeren voor puur de opgetelde vaste kosten, omdat je vermoedelijk met verschillende looptijden zit en daardoor verschillende vaste kosten hebt bij elke contractwissel.

mdr-github, 2021-10-04:
Dank voor je toelichting Dennis. Ik begrijp dat het bedoeld is als indicatie van kosten, geen exacte berekening. En dat het buiten scope/te complex is om alle mogelijke contracten en prijzen te programmeren. Developer tijd is schaars en kan ook anders besteed worden.

Ik ga de workaround met 3 contracten gebruiken: twee voor variabele kosten van gas en electriciteit en een voor de totale vaste kosten. Dat is een slim want wanneer een contract verandert moet je de variabele kosten al aanpassen, de vaste kosten dan ook erbij is dan zo gebeurd. Dit zal voor het een groot deel van de DSMR gebruikers geen probleem zijn, dus als je dit als tip opneemt in de docs ben je misschien al wel ver klaar.