dsmrreader/dsmr-reader #1514: 🙋 InfluxDB met terugwerkende kracht inladen geeft error

Calimerorulez, 2021-11-14:
### Description

Hi,

Ik probeer met terugwerkende kracht de gegevens in een InfluxDB database in te laden, alleen krijg ik de volgende error, als ik de export in de bestaande (waar de huidige metingen in gezet worden) database probeer toe te voegen:

```

2021-11-14 14:40:14,598 INFO     dsmr_influxdb_export_all_readings _export                         140 | INFLUXDB EXPORT: Processing batch (limit: 1000): 878 of 1000
2021-11-14 14:40:14,743 ERROR    dsmr_influxdb_export_all_readings _export_batch                   178 | INFLUXDB EXPORT: Writing measurement(s) failed: 400: {"error":"partial write: field type conflict: input field \"returned_2\" on measurement \"electricity_positions\" is type float, already exists as type integer dropped=28"}
Traceback (most recent call last):
  File "/dsmr/./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.10/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.10/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/usr/local/lib/python3.10/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/usr/local/lib/python3.10/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/dsmr/dsmr_influxdb/management/commands/dsmr_influxdb_export_all_readings.py", line 98, in handle
    self._export(paginator)
  File "/dsmr/dsmr_influxdb/management/commands/dsmr_influxdb_export_all_readings.py", line 146, in _export
    self._export_batch(batch)
  File "/dsmr/dsmr_influxdb/management/commands/dsmr_influxdb_export_all_readings.py", line 176, in _export_batch
    self.influxdb_client.write_points(points)
  File "/usr/local/lib/python3.10/site-packages/influxdb/client.py", line 603, in write_points
    return self._write_points(points=points,
  File "/usr/local/lib/python3.10/site-packages/influxdb/client.py", line 681, in _write_points
    self.write(
  File "/usr/local/lib/python3.10/site-packages/influxdb/client.py", line 413, in write
    self.request(
  File "/usr/local/lib/python3.10/site-packages/influxdb/client.py", line 378, in request
    raise InfluxDBClientError(err_msg, response.status_code)
influxdb.exceptions.InfluxDBClientError: 400: {"error":"partial write: field type conflict: input field \"returned_2\" on measurement \"electricity_positions\" is type float, already exists as type integer dropped=28"}
```

In een losse, aparte database krijg ik de error niet. Het lijkt erop of bij het toevoegen van de historie een dataveld tussentijds veranderd is?

### DSMR-reader version

4.19

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                      v4.19 / v3.10.0 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 0.5s / 168h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             142659
    Consumption records electricity / gas (est.)                                                92201 / 37265
```

dennissiemensma, 2021-11-14:
Bedankt voor je melding. Je kunt wellicht even #1400 doorlezen. Er is toen wel een fix gedaan daarna, maar wellicht dat er toch eenmalig handwerk nodig is. Ik kon dus wel een conversiefout reproduceren en oplossen, maar wellicht niet met terugwerkende kracht.

Calimerorulez, 2021-11-14:
Hi Dennis,

Dank je voor je reactie; ik heb naar aanleiding van #1400 (hoewel ik daar niet helemaal uit kom), een andere strategie van het exporteren/importeren geprobeerd, en dat lijkt te werken.

Ik heb nu eerst de historische data naar een nieuwe Influx database geladen, en vervolgens de live data naar die database laten schrijven. Dan krijg ik de error niet.

Doe ik het andersom (eerst live data laten schrijven om vervolgens het archief eraan toe te voegen), dan werkt het niet :)

closed 2021-11-14 by Calimerorulez

dennissiemensma, 2021-11-15:
Dank voor je aanvulling, fijn dat het gelukt is!