dsmrreader/dsmr-reader #1515: API authorizatie werkt niet?

n9iels, 2021-11-20:
### Description

Ik probeer de DSMR-Reader API te aan te gebruiken, maar wat ik ook doe ik blijft een HTTP 403 terug krijgen. Ik roep bijv. het endpoint `/api/v2/application/monitoring` met de volgende header `Authorization Token xxx`.

Ik snap er niks van. De API key is juist, want deze wordt ook actief gebruik door mijn remote-datalogger en het script daarachter gebruikt op dezelfde manier de Authorization header. Ik snap er niks van 😅 Wat zie ik over het hoofd?

### DSMR-reader version

4.19

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2021-11-20:
Wat krijg je met bijvoorbeeld CURL?

```
curl 'http://localhost/api/v2/datalogger/dsmrreading?ordering=-timestamp&limit=1' \
    -H 'X-AUTHKEY: Test' 
```

```
{"detail":"Invalid auth key"}
```

n9iels, 2021-11-22:
Ik krijg bij beide requests de melding `{"detail":"Invalid auth key"}` terug. Ik heb voor de zekerheid de API key uit mijn remote-datalogger script gekopieert en geplakt. In pinrcippe gebruik ik dus een known-good API key.

closed 2021-11-22 by n9iels

dennissiemensma, 2021-11-22:
Bedankt voor je aanvulling, fijn dat je het zelf hebt kunnen vinden. Dit soort zaken blijven soms lastig te debuggen hoor!